arendst/Tasmota #2890: Watchdog

JIBI3731, 2018-06-02:
Dear fellows,
Thanks to those who made this great tool working on ESP32
I experiment sometimes some situations where the module is just stuck

I suggest to add a feature allowing to define :reboot after xxx minutes without wifi connection, xx minutes without exchanges with the mqtt server. furthermore it shall be possible to program the HW Watchdog to some delay : if not refresh within this delay , reboot
Cheers and thanks
Jerome

Jason2866, 2018-06-02:
If the device is stuck it wont do this things correctly most of the time
because it has to run user code...
My experience is that the devices run a very long time without a problem.
I have many devices online, and not even one got unresponsive or needed a hard reset.
One device worked not as expected-> relay contact was burned out. Was my mistake!

Frogmore42, 2018-06-03:
The esp8266 has a watchdog.  Any embedded system that doesn't have one is asking for problems.  But implementing a really good watchdog is not easy.

Here is a good article on watchdogs and the esp8266.
https://www.sigmdel.ca/michel/program/esp8266/arduino/watchdogs_en.html

There is a software watchdog in Tasmota that will reset the device in certain cases, like no WiFi and probably some others.  This has caused some people issues, since they want it to work independently.

JIBI3731, 2018-06-04:
Thanks @Frogmore42 . however i get a 404 forbidden when clicking your link....
Anyway i'm using the EPS32 port and i actually experiment periodic "stucks" ....
Looking for a solution, HW WD seems to be adequate

Frogmore42, 2018-06-04:
If you are using the esp32 port, you should report it there.  The link works fine for me, but just search watchdog esp8266 and you will likely find it and other sites.

That is one of he problems with watchdogs.  It is hard to get them right, ie reset when things are bad and not when they are okay.  It is both art and science.

ascillato, 2018-06-04:
Hi. Please report the issue in the esp32 fork. Thanks

closed 2018-06-04 by ascillato