arendst/Tasmota #2891: Sonoff POW R2 duplicate mqtt message

andrea-daru, 2018-06-02:
Hi all,
I've flashed the Sonoff POW R2 with Tasmota (self compiled with platformio).
In the main page I see no power value

[image]

In the console I've found that there's always a **duplicate MQTT message** for the current/power value:

```
09:28:51 MQT: tele/ESPPOW/SENSOR = {"Time":"2018-06-02T09:28:51","ENERGY":{"Total":0.027,"Yesterday":0.019,"Today":0.008,"Power":5,"Factor":0.43,"Voltage":218,"Current":0.053}}
09:28:52 MQT: tele/ESPPOW/SENSOR = {"Time":"2018-06-02T09:28:52","ENERGY":{"Total":0.029,"Yesterday":0.019,"Today":0.010,"Power":0,"Factor":0.00,"Voltage":218,"Current":0.000}}
09:29:23 MQT: tele/ESPPOW/SENSOR = {"Time":"2018-06-02T09:29:23","ENERGY":{"Total":0.029,"Yesterday":0.019,"Today":0.010,"Power":5,"Factor":0.44,"Voltage":219,"Current":0.054}}
09:29:24 MQT: tele/ESPPOW/SENSOR = {"Time":"2018-06-02T09:29:24","ENERGY":{"Total":0.029,"Yesterday":0.019,"Today":0.010,"Power":0,"Factor":0.44,"Voltage":219,"Current":0.000}}
09:29:43 MQT: tele/ESPPOW/SENSOR = {"Time":"2018-06-02T09:29:43","ENERGY":{"Total":0.029,"Yesterday":0.019,"Today":0.010,"Power":5,"Factor":0.44,"Voltage":219,"Current":0.052}}
09:29:44 MQT: tele/ESPPOW/SENSOR = {"Time":"2018-06-02T09:29:44","ENERGY":{"Total":0.029,"Yesterday":0.019,"Today":0.010,"Power":0,"Factor":0.00,"Voltage":219,"Current":0.000}}

```
Anoyone else having the same problem? solution?
Thanks
_Andrea_

Jason2866, 2018-06-02:
For help please post status 0

andrea-daru, 2018-06-02:
Hi Jason, can you explain me what is "_Status 0_" ?
Thanks

Jason2866, 2018-06-02:
In web front end hit button "Console"
Type here: Status 0
Post the output from this command

andrea-daru, 2018-06-02:
```
15:46:04 MQT: stat/ESPPOW/STATUS = {"Status":{"Module":43,"FriendlyName":["Sonoff"],"Topic":"ESPPOW","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:46:04 MQT: stat/ESPPOW/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T19:14:46","StartupUTC":"2018-06-01T19:31:18","Sleep":0,"BootCount":3,"SaveCount":20,"SaveAddress":"F5000"}}
15:46:04 MQT: stat/ESPPOW/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:29:54","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:46:04 MQT: stat/ESPPOW/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Daru FRITZ!Box","VodafoneDaru"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
15:46:04 MQT: stat/ESPPOW/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
15:46:04 MQT: stat/ESPPOW/STATUS5 = {"StatusNET":{"Hostname":"ESPPOW-5374","IPAddress":"192.168.1.204","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:F3:EB:B1:D4:FE","Webserver":2,"WifiConfig":3}}
15:46:04 MQT: stat/ESPPOW/STATUS6 = {"StatusMQT":{"MqttHost":"openhabianpi","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B1D4FE","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:46:04 MQT: stat/ESPPOW/STATUS7 = {"StatusTIM":{"UTC":"Sat Jun 02 14:46:04 2018","Local":"Sat Jun 02 15:46:04 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:50","Sunset":"20:46"}}
15:46:04 MQT: stat/ESPPOW/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:46:04 MQT: stat/ESPPOW/STATUS10 = {"StatusSNS":{"Time":"2018-06-02T15:46:04","ENERGY":{"Total":0.030,"Yesterday":0.019,"Today":0.012,"Power":0,"Factor":0.00,"Voltage":220,"Current":0.000}}}
15:46:04 MQT: stat/ESPPOW/STATUS11 = {"StatusSTS":{"Time":"2018-06-02T15:46:04","Uptime":"0T19:14:46","Vcc":3.221,"POWER":"ON","Wifi":{"AP":1,"SSId":"Daru FRITZ!Box","RSSI":76,"APMac":"E0:28:6D:B8:FD:E0"}}}
```

arendst, 2018-06-02:
From status 0 I suggest to disable serial logging as the interface is used by the POW energy monitoring hardware: ``seriallog 0``.

With low loads any powerdelta might report a message so I would disable powerdelta for now and see what happens: ``powerdelta 0``.

As teleperiod is set too 300 (seconds) I cannot explain why you receive messages every 20 seconds...

andrea-daru, 2018-06-02:
With `seriallog 0` and `powerdelta 0` it alwasy log 0 `"Power":0`.
I've also tried to increase the load wothout any differences shown in the log.

arendst, 2018-06-02:
Enable ``weblog 4`` for a few seconds and see if you receive any serial information like:
```
17:30:40 CMD: weblog 4
17:30:40 MQT: stat/pow3/RESULT = {"WebLog":4}
17:30:40 SER: Received 55 5A 02 F7 60 00 03 B1 00 40 10 02 62 DB 51 A6 58 04 02 1E F1 02 8F 91
17:30:40 SER: Received 55 5A 02 F7 60 00 03 B1 00 40 10 02 62 DB 51 A6 58 04 17 21 F1 02 8F A9
17:30:41 SER: Received 55 5A 02 F7 60 00 03 B1 00 40 10 02 62 DB 51 A6 58 04 17 21 F1 02 90 AA
17:30:41 SER: Received 55 5A 02 F7 60 00 03 B1 00 40 10 02 62 DB 51 A6 58 04 17 21 F1 02 90 AA
17:30:41 SER: Received 55 5A 02 F7 60 00 03 B1 00 40 10 02 62 DB 51 A6 58 04 17 21 F1 02 91 AB
17:30:41 SER: Received 55 5A 02 F7 60 00 03 B1 00 40 10 02 62 DB 51 A6 58 04 17 21 F1 02 91 AB
17:30:41 CMD: weblog 2
17:30:41 MQT: stat/pow3/RESULT = {"WebLog":2}
```

andrea-daru, 2018-06-02:
Here it is:
```
16:32:20 MQT: stat/ESPPOW/RESULT = {"WebLog":4}
16:32:20 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 05 CB 8D 52 5D F0 12 59 94 61 00 D8 C9
16:32:20 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 05 CB 8D 52 5D F0 13 F2 F9 61 00 D8 C8
16:32:20 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 05 CB 8D 52 5D F0 15 8C 5E 61 00 D8 C9
16:32:21 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 05 CB 8D 52 5D F0 17 25 C3 61 00 D8 C9
16:32:21 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 05 CB 8D 52 5D F0 18 BF 28 61 00 D8 C9
16:32:21 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 05 CB 8D 52 5D F0 1A 58 8D 61 00 D8 C9
16:32:21 CFG: Saved to flash at F7, Count 34, Bytes 2560
16:32:21 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 05 CB 8D 52 5D F0 1B F1 F2 61 00 D8 C8
16:32:21 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 05 CB 8D 52 5D F0 1D 8A 58 61 00 D8 C9
16:32:21 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 1D 8A 58 61 00 D8 F6
16:32:21 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 1D 8A 58 61 00 D8 F6
16:32:21 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 1D 8A 58 61 00 D8 F6
16:32:21 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 1D 8A 58 61 00 D8 F6
16:32:21 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 1D 8A 58 61 00 D8 F6
16:32:22 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 1D 8A 58 61 00 D8 F6
16:32:22 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 1D 8A 58 61 00 D8 F6
16:32:22 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 1D 8A 58 61 00 D8 F6
16:32:22 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 1D 8A 58 61 00 D9 F7
16:32:22 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 0F D4 09 61 00 D9 E4
16:32:22 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 11 6D 6E 61 00 D9 E4
16:32:22 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 13 07 D2 61 00 D9 E4
16:32:22 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 14 A1 37 61 00 D9 E4
16:32:22 SER: Received 55 5A 02 DA 78 00 03 5D 00 40 A1 06 7C 08 52 5D F0 16 3A 9C 61 00 D9 E4
16:32:28 SER: Received 55 5A 02 DA 78 00 03 5C 00 40 A1 05 D1 32 52 5D F0 1D 60 75 71 00 DB 79
16:32:28 SER: Received 55 5A 02 DA 78 00 03 5C 00 40 A1 05 D1 32 52 5D F0 1D 60 75 71 00 DC 7A
16:32:28 SER: Received 55 5A 02 DA 78 00 03 5C 00 40 A1 05 D1 32 52 5D F0 10 93 5C 61 00 DC 77
16:32:28 SER: Received 55 5A 02 DA 78 00 03 5C 00 40 A1 05 D1 32 52 5D F0 12 2C C1 61 00 DC 77
16:32:28 SER: Received 55 5A 02 DA 78 00 03 5C 00 40 A1 05 D1 32 52 5D F0 13 C6 26 61 00 DC 77
16:32:28 SER: Received 55 5A 02 DA 78 00 03 5C 00 40 A1 05 D1 32 52 5D F0 15 5F 8B 61 00 DC 77
16:32:28 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 16 F8 F0 61 00 DC 78
16:32:28 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 18 92 55 61 00 DC 79
16:32:28 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1A 2C B9 61 00 DC 79
16:32:29 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1B C5 1F 61 00 DC 79
16:32:29 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1D 5F 83 61 00 DC 79
16:32:29 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1E 43 BD 61 00 DC 98
16:32:29 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1E 43 BD 61 00 DC 98
16:32:29 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1E 43 BD 61 00 DC 98
16:32:29 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1E 43 BD 61 00 DC 98
16:32:29 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1E 43 BD 61 00 DC 98
16:32:29 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1E 43 BD 61 00 DC 98
16:32:29 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1E 43 BD 61 00 DC 98
16:32:29 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1E 43 BD 61 00 DC 98
16:32:30 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 05 D1 32 52 5D F0 1E 43 BD 61 00 DC 98
16:32:30 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 1E 43 BD 61 00 DD 11
16:32:30 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 10 B2 5C 61 00 DD 11
16:32:30 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 12 4B C1 61 00 DD 11
16:32:30 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 13 E6 25 61 00 DD 11
16:32:30 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 15 7F 8A 61 00 DD 11
16:32:30 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 17 17 F0 61 00 DD 11
16:32:30 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 18 B1 55 61 00 DD 11
16:32:30 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 1A 4A BA 61 00 DD 11
16:32:31 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 1B E4 1F 61 00 DD 11
16:32:31 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 1D 7D 84 61 00 DD 11
16:32:31 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 1E 71 4E 71 00 DD E0
16:32:31 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 1E 71 4E 71 00 DD E0
16:32:31 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 1E 71 4E 71 00 DD E0
16:32:31 SER: Received 55 5A 02 DA 78 00 03 5E 00 40 A1 06 A6 D4 52 5D F0 1E 71 4E 71 00 DD E0
[... cut ...]
16:32:43 SER: Received 55 5A 02 DA 78 00 03 5F 00 40 A1 04 F7 1C 52 5D F0 20 F1 73 71 00 E4 26
16:32:43 SER: Received 55 5A 02 DA 78 00 03 5F 00 40 A1 04 F7 1C 52 5D F0 10 C0 2F 61 00 E4 91
16:32:44 SER: Received 55 5A 02 DA 78 00 03 5F 00 40 A1 04 F7 1C 52 5D F0 12 59 94 61 00 E4 91
16:32:44 SER: Received 55 5A 02 DA 78 00 03 5F 00 40 A1 04 F7 1C 52 5D F0 13 F2 F9 61 00 E4 90
16:32:44 SER: Received 55 5A 02 DA 78 00 03 5F 00 40 A1 04 F7 1C 52 5D F0 15 8C 5E 61 00 E4 91
16:32:44 SER: Received 55 5A 02 DA 78 00 03 62 00 40 A1 04 F7 1C 52 5D F0 17 25 C3 61 00 E4 94
```

arendst, 2018-06-02:
Looking at the CF count (byte 21and 22 being E2, E3, E4) there seems to be a very low power load connected. This is also represented in many unmeasurable power loads indicated by byte 20 which should say 71 when power, voltage and current could be measured. I see many 61 = invalid power meaning to power load is too low to measure.

Suggestions:
- Add a load of at least 20W like an old incandescent bulb which also is a full resistive load.
- Calibratie voltage, current and power using this bulb.

andrea-daru, 2018-06-02:
Ok you're right... the load was minimal (5w)... I've attached a 60w light bulb and now it work!
Thank you for the support :)

closed 2018-06-02 by andrea-daru