arendst/Tasmota #2893: Sonoff Basic R2 DO NOT WORK with Tasmota

wfdudley, 2018-06-02:
I have a dozen Sonoff Basic, and they all work perfectly with 5.12.0e.

I recently got two Sonoff Basic R2, and they DO NOT WORK with the following releases:

5.10.0
5.12.0e
5.14.0b

"Does Not Work" is defined as: cannot connect to wi-fi, or, if they can connect to wi-fi, cannot connect to the mosquitto server.  This is gleaned from watching the serial output on the arduino monitor window.

I'm using the latest Arduino 1.8.5 to build and upload.  I'm using devuan Linux (almost the same as debian).  I'm an experienced developer who admins FreeBSD and Linux systems (as a hobby, since I'm retired).  I have many completed Arduino projects, some quite complicated.

Someone suggested that I use Platformio, because <mumble library something something>, and that has been an unmitigated train wreck.  There are essentially no instructions on how to get that to work, Platformio keeps throwing errors without suggestions on how to fix them, google searches have turned up nothing useful, and so that's just been a waste of several hours of my life.  In particular, the instructions on the Tasmota wiki are useless unless you're already an expert and have done this before.

I will PAY someone to give me CLEAR, CONCISE instructions on how to program Sonoff Basic R2 using anything that will run on debian Linux (amd64), but preferably Arduino.  You get paid when the build/upload process completes without errors.  I am serious.  I want to keep using Sonoffs, and I can't if all future production units are broken and won't work with Tasmota.

wfdudley, 2018-06-02:
> Hello Dudly
> have you seen my last entrys in #1973

Re: your entries in bug 1973:

The platformio tutorial is broken, as platformio throws errors when I try
to follow those instructions.
I'd copy them into a message, but the error messages evaporate and are
therefore non-copyable.

Honestly, I'm done with platformio, it's horribly broken.  Life is too
short to spend my life
fixing the problems with that program.

Re: the Arduino tutorial: I've already used the Arduino platform to
successfully program
15 Sonoff's, so I know how to use Arduino well enough.

It's only the R2s that are broken.

Bill Dudley

This email is free of malware because I run Linux.

arendst, 2018-06-02:
Let's go back to your favourite IDE.

Pls provide a screenshot of your compile parameters.

wfdudley, 2018-06-02:
I assume by compile parameters, you mean "what you see when you click 'Tools'", which is attached.
[image]

arendst, 2018-06-02:
Thnx. Looks totally Ok.

Could you try to catch the output of serial command status 0

wfdudley, 2018-06-02:
I tried sending "status 0", "0", and "command status 0" and got no interesting output.  Here is what I am seeing on the serial monitor:

`00:00:21 WIF: Connect failed with AP timeout
00:00:21 WIF: Connecting to AP2 Yeldud UP in mode 11N as sonoff-3018...
00:00:42 WIF: Connect failed with AP timeout
00:00:42 WIF: WPSConfig active for 3 minutes
00:01:03 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_A7CBCA, GroupTopic sonoffs) Version 5.14.0b-2_4_1
00:00:00 WIF: Connecting to AP2 Yeldud UP in mode 11N as sonoff-3018...
00:00:12 WIF: Connect failed with AP incorrect password
00:00:12 WIF: Connecting to AP1 Yeldud's Bar in mode 11N as sonoff-3018...
00:00:32 WIF: Connect failed with AP timeout
00:00:32 WIF: WPSConfig active for 3 minutes
`

wfdudley, 2018-06-02:
That's a dog's breakfast, isn't it.  I'll try to format that.  wait one sec . . .

wfdudley, 2018-06-02:
00:00:10 CMD: seriallog 3
00:00:10 RSL: stat/sonoff/RESULT = {"SerialLog":"3 (Active 3)"}
00:00:10 CFG: Saved to flash at FB, Count 42, Bytes 3584
00:00:11 WIF: Attempting connection...
00:00:12 WIF: Attempting connection...
00:00:12 CMD: status 0
00:00:12 SRC: Serial
00:00:12 RSL: Group 0, Index 1, Command STATUS, Data 0
00:00:12 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:12 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:11","StartupUTC":"","Sleep":0,"BootCount":20,"SaveCount":42,"SaveAddress":"FB000"}}
00:00:12 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.14.0b","BuildDateTime":"2018-06-02T00:12:43","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
00:00:12 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Yeldud's Bar","Yeldud UP"],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
00:00:12 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":554,"Free":448,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","17B6179A","00000000"]}}
00:00:12 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3018","IPAddress":"0.0.0.0","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"84:F3:EB:A7:CB:CA","Webserver":2,"WifiConfig":3}}
00:00:12 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"mail.casano.com","MqttPort":16947,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A7CBCA","MqttUser":"elrey1848","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:12 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:12 1970","Local":"Thu Jan 01 00:00:12 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:00:12 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:12"}}
00:00:12 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:12","Uptime":"0T00:00:11","Vcc":3.316,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Yeldud's Bar","RSSI":100,"APMac":"74:DA:38:D5:C0:3B"}}}
00:00:13 WIF: Attempting connection...
00:00:14 WIF: Attempting connection...
00:00:15 WIF: Attempting connection...
00:00:16 WIF: Attempting connection...
00:00:17 WIF: Attempting connection...
00:00:18 WIF: Attempting connection...
00:00:19 WIF: Attempting connection...
00:00:20 WIF: Attempting connection...
00:00:21 WIF: Connect failed with AP timeout
00:00:21 WIF: Connecting to AP2 Yeldud UP in mode 11N as sonoff-3018...
00:00:22 CFG: Saved to flash at FA, Count 43, Bytes 3584
00:00:22 WIF: Attempting connection...
00:00:23 WIF: Attempting connection...
00:00:24 WIF: Attempting connection...
00:00:25 WIF: Attempting connection...
00:00:26 WIF: Attempting connection...
00:00:28 WIF: Attempting connection...
00:00:29 WIF: Attempting connection...
00:00:30 WIF: Attempting connection...
00:00:31 WIF: Attempting connection...
00:00:32 WIF: Attempting connection...
00:00:33 WIF: Attempting connection...
00:00:34 WIF: Attempting connection...
00:00:35 WIF: Attempting connection...
00:00:36 WIF: Attempting connection...
00:00:37 WIF: Attempting connection...
00:00:38 WIF: Attempting connection...
00:00:39 WIF: Attempting connection...
00:00:40 WIF: Attempting connection...
00:00:41 WIF: Attempting connection...
00:00:42 WIF: Connect failed with AP timeout
00:00:42 WIF: WPSConfig active for 3 minutes

arendst, 2018-06-02:
Again, looks great.

What happens if you enter the correct password for Yeldud Up?

command password1 xxxxx

Compare the output with a functional sonoff and look for possible differences.

Jason2866, 2018-06-02:
I think the problem is "Core":"2_4_1","SDK":"2.2.1
Try as i wrote in the other thread Core 2.3.0

wfdudley, 2018-06-02:
A search of Sonoff-Tasmota issues using the term "Core 2.3.0" finds 29 open and 51 closed.
Could you be more specific about what issue I should read?  Or at least give me a better
search term?

Thanks

arendst, 2018-06-02:
Dont go that way already. I suppose your current sonoffs run 2.4.1 too right? Well at least I run 2.4.1 on dozens of sonoffs for some weeks now without any problem. So for now stick to your current 2.4.1

I suggest you play around with the available commands and compare a working sonoff with the R2.

Options are also

reset 1
reset 2

BTW I do not own a R2 so cannot verify.

Jason2866, 2018-06-02:
@arendst 
I have some sonoff basics R1 and two R2. I can confirm that R1 works well with 2.4.1.
One of my two R2 doesnt work with 2.4.1. with 2.3.0 it works well too.
I have no idea what is the difference. I spent a plenty of time to find the reason.
I havent really found a answer. I think the use different charges of the ESP Chip.

wfdudley, 2018-06-02:
I don't have any idea to what "2.4.1" refers, but all the other 20 or so Sonoff's run 5.12.0, are compiled and flashed with the box stock Arduino 1.8.5 environment, using the instructions here: https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

The R2's mostly time out, except when the partially get through the auth sequence, in which case they report "incorrect password".  Actually, one of the R2s fails at WiFi all of the time, and the other one usually gets a wifi connection, and then has extreme difficulty keeping a connection to the mosquitto server.  It "sort of" works, except it's constantly having to re-try the connection.

I will happily mail you BOTH of the R2s so you can try to figure out what's wrong.  I don't know enough about /your code/wifi/low level networking/ to be able to do any useful debugging.  And since the units are broken (and only $6 each) I can afford to lose them forever if it will fix the problem for *everybody* going forward.

By the way, Arduino is hardly "my favorite IDE" (that's vi and make in two xterms) but the Arduino IDE works and is fairly reliable.

Jason2866, 2018-06-02:
You find it here (Boardmanager):
[image]

wfdudley, 2018-06-02:
@Jason2866, that's the first time I've understood what you were writing about, with no ambiguity and very little head scratching.

My Arduino environment has 2.4.1 installed.

One mystery solved.  Know I understand your first message, from like 10 messages ago in the *other* thread.  Pity I wasted all that time with atom/whatever it's called.

Bill Dudley

wfdudley, 2018-06-02:
Changing from 2.4.1 to 2.3.0 causes the Arduino compile to fail.  "Reliably" == "twice in a row"

Console output from Arduino IDE:

Picked up JAVA_TOOL_OPTIONS: 
Archiving built core (caching) in: /tmp/arduino_cache_923507/core/core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_dout,UploadSpeed_115200,FlashSize_512K64,ResetMethod_nodemcu,Debug_Disabled,DebugLevel_None_____91c5c5378d343ae609187fa24bbc46a0.a
/home/dud/arduino-1.8.5-sonoff-5.14/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sonoff.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.

Bill Dudley

wfdudley, 2018-06-02:
I downgraded to 2.3.0, and that broke compiling.

Picked up JAVA_TOOL_OPTIONS: 
Archiving built core (caching) in: /tmp/arduino_cache_923507/core/core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_dout,UploadSpeed_115200,FlashSize_512K64,ResetMethod_nodemcu,Debug_Disabled,DebugLevel_None_____91c5c5378d343ae609187fa24bbc46a0.a
/home/dud/arduino-1.8.5-sonoff-5.14/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sonoff.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.

Installing 2.3.0 changed flash size to "512K (64K SPIFFS)", and there is no option for
"1M (no SPIFFs)".

So that was a fail.

Bill Dudley

arendst, 2018-06-03:
The correct linker for 1M(no spiffs) is provided with the tasmota package and needs to be installed as described in the wiki when 2.3.0 is being used.

wfdudley, 2018-06-03:
I'll try that when next I have a chance, which will be in three weeks.

Bill Dudley

This email is free of malware because I run Linux.

joba-1, 2018-06-04:
@wfdudley: wont solve your problem here, but since you also like vi/make: you may give up on atom, but stay a little longer with platformio: you can build and upload from the shell (pio command), very versatile - I love it.

autofabrik, 2018-06-24:
I’m having the same issues here. I’ve built 6.0.0a and it flashes fine with Platform io. Won’t connect to predefined AP and won’t broadcast a network when button pressed 4 times. All serial outputs concur with this but no WiFi at all it seems. Tried WPS too no luck. Also built and tried 5.1.4 with no luck.

Jason2866, 2018-06-24:
Do you build your versions with 2.3.0? All! my devices works with that library and every Tasmota version including 6.0.a

autofabrik, 2018-06-25:
Jason8266. I tried building 6.0.0a with 2.3.0 no luck I'm afraid. Will try again later if I get some time.

autofabrik, 2018-07-01:
Success. Used a new desktop machine. Fresh install of MSVSCode / PlatformIO and new clone of the latest build. Left the setting as 2.3.0 (platform = espressif8266@1.5.0) and success worked first try. Will now see about stability etc. I see there seems to be an issue on the espressif8266 GitHub about platformio linking problems with later versions.

wfdudley, 2018-07-04:
Since my last post, I've been travelling.  I had done the earlier work at my brother's home, and now I'm at my home.  I have one of his R2 devices here to play with.  Today I did the following:

1. modified user_config.h to have MY WiFi SSID and password, instead of my brother's SSID & PW.
2. Brought up the Arduino IDE and did a compile/load of this code (version 5.12.0e and ESP8266 2.4.1).
3. When the upload completed, I brought up the Arduino serial monitor.

Rather surprisingly, the debug stream indicated that the Sonoff was attempting to connect to WiFi
with the OLD SSID and password.  I have no idea how it could be using the SSID and password from
my brother's house, as I'd changed those parameters in user_config.h.

Since the Sonoff was failing to connect to WiFi (wrong SSID and PW) I held down the button on the
Sonoff for several seconds, with the idea that I'd get it into access point mode and fix the SSID and
password.

Surprisingly, that "reset" the Sonoff, and it immediately connected to my WiFi (obviously using the
correct SSID and password).

I have NO IDEA why this worked.  Can anyone explain this to me?  I never had to hold down the button
on the R1 Sonoffs to get them to work.  They just worked immediately after flashing.

Thanks,
Bill Dudley

Jason2866, 2018-07-04:
Flashing a new Tasmota firmware doesnt delete settings on the device. Wifi and all other settings survive a flash. With a long key press the device is resetted to the values stored in user_config.h

phathak, 2018-08-11:
Hey,
Having the same issue with a Sonoff basic R2. Getting the same output...
`00:00:30 WIF: Connect failed with AP incorrect password`
I'm using the master branch. Is the dev branch supporting R2s?

Thank

phathak, 2018-08-11:
Well it worked. For people with this issue, if you take a look at the button functionality in the wiki it says : 

> Pressing the button for over forty seconds: Reset settings to defaults as defined in user_config.h and restarts the device.
> 
It was true that the password was wrong the first time I set it in the config (had a space at the end). But reflashing it wasn't erasing the old password. Pressing the button for forty seconds or more will make the light stop blinking and reset the config using the one provided in the user_config; hence the corrected password.

Jason2866, 2018-08-11:
Maybe a corrupt setting is left in device...
Erase the device with esptool. After that flash Tasmota. 
Always helped me with devices that had strange behaviour

ascillato2, 2018-08-22:
Hi,

Closing this issue as wfdudley has stated that the issue is solved and the questions has been answered. Please, ask to reopen if needed. Thanks.

closed 2018-08-22 by ascillato2