arendst/Tasmota #2896: Sonoff Basic R2 V1.0 2017-10-11 1750 Flash problem

mheilemann, 2018-06-02:
Sonoff Basic R2 V1.0 2017-10-11 1750 Flash problem on Windows 10, using Arduino 1.8.5.

I have tried it with both the ESP8266 2.3.0 and 2.4.1 but always get:

Archiving built core (caching) in: C:\Users\Hammer\AppData\Local\Temp\arduino_cache_603806\core\core_esp8266_esp8266_generic_UploadTool_esptool,CpuFrequency_80,FlashFreq_40,FlashMode_dout,UploadSpeed_115200,FlashSize_1M0,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____3436dab4bbf3d6c5facd37dbacf5bcc6.a
Sketch uses 537332 bytes (52%) of program storage space. Maximum is 1023984 bytes.
Global variables use 56508 bytes (68%) of dynamic memory, leaving 25412 bytes for local variables. Maximum is 81920 bytes.
**warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed**

I have followed the tutorial here to a T: https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE
I have tried both tx-rx, rx-tx and tx-tx rx-rx with the same results on 2 different sonoff modules.

I am trying to use an FTDI232 purchased off amazon: https://www.amazon.com/gp/product/B00IJXZQ7C/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1

The lights look like they are trying to communicate with the board (and yes it is in flash mode after powering it up with the button down...) then gives the error above.

Any help would be appreciated...

reloxx13, 2018-06-02:
Did you change to 3,3V on the FTDI?

Always 
RX <-> TX
TX <-> RX

Guess you are not pressing the Button correct or using a wrong only for charging USB Cable and dont use a USB Hub between.

mheilemann, 2018-06-02:
Yes to the above, the FTDI is on 3.3 and I have tried it both ways just to make sure.  The green light flashes and I get signals on the FTDI led's when it is not in flash mode so I am sure that isn't the problem...

I have hooked the USB to multiple different ports, with no hub.

reloxx13, 2018-06-02:
Is COM5 (Default in WIki) correct?

You can check this in the Device Manager from Windows.

mheilemann, 2018-06-02:
It is on Com 4 for me, I haven't been able to get it to show up on any other p
[image]
ort

reloxx13, 2018-06-02:
It doesnt matter as long you choose it in Arduino IDE.

Okay, last thing the Jumper Wires. They r cheap made and it happens quite often that they are not crimped correct and are not connecting.

reloxx13, 2018-06-02:
In platformio sometimes i have to wait a lil bit more before inserting the USB in the PC after pressing on Upload

My steps are:
Clean
Compile
Upload
Wait a lil bit
Plugin USB with Btn on Sonoff pressed.

mheilemann, 2018-06-02:
I tested all the connections with my ohm meter and they are connected.  I am trying the upload with the Arduino IDE per the instructions, and have not tried using platformio.

reloxx13, 2018-06-02:
Did you solder the Pins in and you are 120% sure they are soldered in correctly?

Im getting out of ideas since you did everything right till now 👍

Jason2866, 2018-06-02:
I dont like FTDI devices for flashing sonoffs. The often deliver not enough current on the 3.3V path.
Second most of them are fake ones...
I have made good experiences with USB - serial adapters with a CH340 or CH341 chip.

reloxx13, 2018-06-02:
Drivers i use:
http://www.ftdichip.com/Drivers/D2XX.htm

Its a newer version as from windows servers which gets installed automatic.

but same, i dont think this is the issue, since the leds are flashing.

Jason2866, 2018-06-02:
Or try Visual Studio Code ;-) I like it a little more than PlatformIO.
Both a lightyears better than Arduino for Tasmota

DynamoEffects, 2018-06-05:
If it helps, I had the exact same problem with a 4ch Pro R2 and it was because it hadn't entered programming mode.  Make absolutely sure that GPIO1 and GND are connected for at least 3 seconds when power is applied.