arendst/Tasmota #2899: sonoff basic cannot re-flash after failed flash

kbaras, 2018-06-03:
Hello,

I have following issue: I tried to flash Tasmota on a stock sonoff basic device. It said "started flashing", but just after "failed, no answer received" or something similar.
Since then I am unable to do anything with it. Unable to re-flash (like it does not get into flash mode, or cannot communicate to the esp).

Serial output is totally unreadable:
ř[00]ü[00][00]

Any hints on that? 
Thank you!

Jason2866, 2018-06-03:
If you flashed with a 3.3V Usb serial adapter i dont think it is defect. Flash simply failed.
Just flash it again. Press and hold button before you connect the USB adapter to the PC.
Many problems are a result of not powering the sonoff with enough current on the 3.3V path.
Use a proper 3.3V power supply. FTDI adapters are in general NOT a good choice for that...
I have good results using a adapter wirh a CH340 chip. I dont need here a external power supply.

Jason2866, 2018-06-03:
Same as #2896

kbaras, 2018-06-03:
Thank you for reply.
It must have another cause as I can use this ftdi to flash other sonoffs without any issues. I have just took another sonoff basic and was able to re-flash it, but this piece (sonoff basic) has still the same problem.

Frogmore42, 2018-06-03:
Flashing can not work for many reasons.  Making it work is a matter of eliminating the things it might be till you get to the one or more things it is.

Many people have had trouble flashing when using an FTDI to power their esp8266.  If it has worked for you on other ones, consider yourself lucky, since it does not always work.  The FTDI is not designed to supply the amount of current that an esp8266 needs.

Another thing it could be is that your Sonoff is defective.  I had one of those.  Even when powered properly, I could not flash it.  I finally gave up on it and used a different one.  Since they are cheap, that was the easy solution.

If you have a good oscilloscope (and know how to use it) you can measure VCC over time to see if the voltage on the line drops below about 3V when attempting to flash.  That will give you a good picture of just how well the FTDI can supply power.  I have not done this while flashing, but I have done it on a NodeMCU under normal use.  Even with the bigger regulator it has (rated for 800mA, vs the 50mA on the FTDI) it frequently gets down to 3V(for short periods).

So, try it with providing proper power.  If it still isn't working, your Sonoff is dead, buy a new one.

ascillato, 2018-06-12:
Hi, if your question was addressed, please close this issue. Thanks :+1:

closed 2018-06-27 by ascillato2