arendst/Tasmota #2900: Help flashing TYWE3S

galak, 2018-06-03:
I picked up one of the following dimmer switches:

https://www.amazon.com/gp/product/B07CTNSZZ8/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

It's got a TYWE3S module in it, I was able to get connected and see debug output from the uart as well as getting it into what seems boot mode.  However, whenever I try esptool to flash nothing happens.

Not sure if this is the best place to ask for help in trying to figure out what's going on.

Frogmore42, 2018-06-03:
Search the issues for esptool, there are literally over a hundred issues.  It is unlikely that you have a new unique issue.  There are many reasons why flashing fails.  It is also unclear what "nothing happens" means.  All tools provide some output indicating what went wrong.  Search the issues for the message you got and you will see some of the things others did to solve their problems.

galak, 2018-06-04:
> Search the issues for esptool, there are literally over a hundred issues. It is unlikely that you have a new unique issue. There are many reasons why flashing fails. It is also unclear what "nothing [...]

Thanks, I've tried many of those things.  What I'm seeing is:

```
$ esptool.py --baud 9600 --port /dev/tty.usbserial-00000000 read_flash 0x00000 0x100000 dimmer.bin
esptool.py v2.3.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```

I think this is related to the fact that the UART on the EPS8266 is connected to some other chips.

Frogmore42, 2018-06-04:
Are you sure you have selected the correct port?  It seems unlikely it would the on you tried.
The error indicates one of these things:

1. You are using the wrong port
1. Your device is defective
1. Your device is not getting good enough power
1. You don't have it wired correctly 

I don't use Linux regularly, so I don't know how you figure out which port to use.  This is easy on Windows and with PlatformIO it figures it out for you.
I always use a separate 3V3 regulator to power the esp8266 when flashing
There are only two ways to hook up the serial wires.  I use Teraterm to validate the wires are connected.

What do you mean the UART is connected to other chips?  If you mean the serial lines on the tywe3s are wired to stuff on the board, that might cause a problem.  But, I would expect that they flash it in the factory after it is soldered together.

Have you flashed other esp8266 based devices with the same setup?

galak, 2018-06-04:
> Are you sure you have selected the correct port? It seems unlikely it would the on you tried.
> The error indicates one of these things:
> 
> You are using the wrong port
> Your device is defective
> Your device is not getting good enough power
> You don't have it wired correctly
> I don't use Linux regularly, so I don't know how you figure out which port to use. This is easy on Windows and with PlatformIO it figures it out for you.
> I always use a separate 3V3 regulator to power the esp8266 when flashing
> There are only two ways to hook up the serial wires. I use Teraterm to validate the wires are connected.
> 
> What do you mean the UART is connected to other chips? If you mean the serial lines on the tywe3s are wired to stuff on the board, that might cause a problem. But, I would expect that they flash it in the factory after it is soldered together.
> 
> Have you flashed other esp8266 based devices with the same setup?

I'm able to get output from the the ESP8266, so I would assume that would rule out 1-3.

> What do you mean the UART is connected to other chips? If you mean the serial lines on the tywe3s are wired to stuff on the board, that might cause a problem. But, I would expect that they flash it in the factory after it is soldered together.

It appears that there is a microcontroller and cap touch sensor chip.  I'm not exactly sure how they are wired together, since it seems that the microcontroller & cap touch sensor are both connected to the uart -- which doesn't make any sense to me.

I've tried removing the microcontroller and that hasn't helped.  I need to try to remove the cap sensor, but that's more involved.

> Have you flashed other esp8266 based devices with the same setup?

I have, so I'm guessing the issue is the other devices connected to the uart.

I posted so pictures here #469

Frogmore42, 2018-06-04:
Looking at the other issue, I believe you are correct.  But, even if you get it flashed, I doubt that it will work with Tasmota.  You would need to decode the communication between the ESP and the other MCU and then implement that in the Tasmota codename.  So you should do that work now before you lose the ability to decode it by flashing Tasmota.

galak, 2018-06-04:
> Looking at the other issue, I believe you are correct. But, even if you get it flashed, I doubt that it will work with Tasmota. You would need to decode the communication between the ESP and the [...]

Agreed, I wanted to start with just dumping the flash that is there now -- to make sure I'm able to connect and possibly write a new binary, before I invest a lot of time doing other things.

digiblur, 2018-08-15:
@galak have you had any further luck?  I was able to dump the firmware without issue on the dimmer.  Looked through with a bit of a hex editor.  Trying to figure out how they are talking to that other chip over the RX/TX lines and haven't had much luck with baud rates and such.

twonky26, 2019-12-21:
The method that worked for me was cutting RX and TX pins on BF7612AM28T chip which is located on the the other side of the board (of Moes DS01C dimmer ). So I guess it will work for any device with TYWE3S which has trouble flashing. 

First, trace RX and TX pins from TYWE3S to BF7612AM28T (or similar) chip.
Second, carefully cut them with sharp blade.
Third, bend them up a little bit, so there is no contact.
After that it should flash just fine. But don't forget to bend them down and resolder after flashing.

[image]
[image]
[image]
[image]
[image]
[image]

joshgarde, 2020-05-22:
Just in case anyone comes across this thread in the future, I had success flashing a Tuya device with a UART device attached to the TYWE3S by driving the reset pin on the secondary MCU low with a jumper wire. No pin cutting or desoldering required

vhandrade, 2020-09-01:
joshgarde Which pins did you connect? I'm experiencing this issue with some TreatLife switches.

joshgarde, 2020-09-01:
> joshgarde Which pins did you connect? I'm experiencing this issue with some TreatLife switches.

Depends on which secondary MCU you have. I used the part number on my MCU to get a datasheet and locate the reset pin. From there I found a ground pad and then soldered a jumper wire. Fiddly setup, but it was enough to get the TYWE3S flashed.