arendst/Tasmota #2901: [5.14.0] web athentication fails if credentials aren't part of the url string

j-ed, 2018-06-03:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used :  OBI Wifi socket

I've installed the latest Tasmota image v5.14.0 on my wifi socket and all functions are working as  expected, except that the authentication credentials are only accepted if they're parsed via the http url string, like:
```
# curl "http://192.168.1.130/cm?user=admin&password=password&cmnd=Power%20Status"
*   Trying 192.168.1.130...
* TCP_NODELAY set
* Connected to 192.168.1.130 (192.168.1.130) port 80 (#0)
> GET /cm?user=admin&password=password&cmnd=Power%20Status HTTP/1.1
> Host: 192.168.1.130
> User-Agent: curl/7.59.0
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Type: application/json
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: -1
< Content-Length: 14
< Connection: close
< Access-Control-Allow-Origin: *
<
* Closing connection 0
{"POWER":"ON"}
```
If I use an external password file for the credentials (`machine 192.168.1.130 login admin password password`), to increase security, the authentication fails with the following warning:
**Benötige user=<< Benutzername >>&password=<< Passwort >>**
```
# curl -vvvv --netrc-file ./aaa-tasmota.pw 'http://192.168.1.130/cm?cmnd=Power%20Status'

*   Trying 192.168.1.130...
* TCP_NODELAY set
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0* Connected to 192.168.1.130 (192.168.1.130) port 80 (#0)
* Server auth using Basic with user 'admin'
> GET /cm?cmnd=Power%20Status HTTP/1.1
> Host: 192.168.1.130
> Authorization: Basic YWRtaW46c29ub2ZmMjAxOA==
> User-Agent: curl/7.59.0
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Type: application/json
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: -1
< Content-Length: 63
< Connection: close
< Access-Control-Allow-Origin: *
<
{ [63 bytes data]
100    63  100    63    0     0     35      0  0:00:01  0:00:01 --:--:--    35
* Closing connection 0
WARNING":"Benötige user=<Benutzername>&password=<Passwort>"
```
If I understood the change notes correctly, basic authentication should be supported since v3.9.20 (https://github.com/arendst/Sonoff-Tasmota/pull/87). Can anyone please explain to me why it doesn't work to authenticate that way?

arendst, 2018-06-04:
What's in a name.

If curl composes a url using your credentials file like
```
http://admin:password@192.168.1.130/cm?cmnd=Power%20Status
```
it will fail as Tasmota only recognizes what you have noticed:
```
http://192.168.1.130/cm?user=admin&password=password&cmnd=Power%20Status
```
for the simple reason that the ESP8266/Arduino core version 2.3.0 did not support the "official" way of authentication.

To stay backward compatible this is still the case for Tasmota using core 2.4.1

j-ed, 2018-06-04:
@arendst Thank you for the confirmation, I wasn't sure if this is an unwanted behavior or not.
Do you have any plans to implement that "official" authentication mechanism in a future release?

arendst, 2018-06-04:
Once there is a stable core version without current issues like lock-up on sleep or wifi disconnect memory leaks allowing to abandon 2.3.0 I will reconsider.

j-ed, 2018-06-04:
Ok, should I leave this issue ticket open or should I close it for the meantime?

arendst, 2018-06-04:
Pls close it

closed 2018-06-04 by j-ed