arendst/Tasmota #2902: Sonoff Basic cannot boot with GPIO 01 (TX) LOW

alfredopironti, 2018-06-03:
I have a sonoff basic to which I have attached switches to the following GPIOs: 01 (TX), 03 (RX), 14.

All above GPIOs are pulled up with a 10K resistor. When each switch closes, the respective GPIO is shorted to GND.

The sonoff correctly boots regardless of the status of GPIOs 03 and 14 (switch open, hence GPIO pulled up; or switch closed, hence GPIO connected to GND).

However, the sonoff does not boot if GPIO 01 is LOW (connected to GND, with the switch closed).

I'm not sure whether this is an ESP8266 issue, or a Tasmota one. However, I haven't found much documentation about this around, and wanted to share my experience here. Does anyone else experienced this issue? If it's an hardware design, it could be documented in the wiki, so that one can take this constraint into account when designing a system.

Jason2866, 2018-06-04:
Serial GPIO 1 with 0 Ohm connected at GND at boot time results in stuck device.
This behaviour is normal (not documented?, or i didnt find it...).
@Frogmore42  you cant disable serial function at boot moment, only at a later time

joba-1, 2018-06-04:
I think, this is what happens: when serial output pulls tx high you basically short cirquit it through your switch to ground. Voltage probably goes below 3V at that point.
Try a high ohm pullup to high and a lower ohm pulldown on the switch, like 20k and 2k

TOBYBEAR69, 2018-06-09:
What version of Tasmota are you using?  I had a similar problem with 6.0.0a today.  I was trying to hook up a second touch switch (never tried that with a Sonoff Basic before) which requires a 10K pull down resistor.  Normally I only need one so I just connect it to GPIO 14.  When I added the second to GPIO 1 it would not boot.  I'm not 100% sure this is what fixed my problem....but it working fine with 5.14.0.a.  Again, that is with a 10K pull down resistor on GPIO 1.  I'll test my final working setup on 6.0.0a to confirm.  I agree with @Jason2866 ....disabling Serial Logging does not work....tried that early on and it did not help.  It has always been my understanding that the only restrictions are: GPIO 0 and 2 must float on boot and GPIO  0 and 15 must be pulled down on flash.  I have used TX and RX for years without problem but mainly on WEMO D1 Mini's.  Their serial RX and TX pins may have less restrictions because you can use USB for that.

Jason2866, 2018-06-09:
In general GPIO 0 and 2 should be pulled up for reliable working.
It is a must if you connect switches!

TOBYBEAR69, 2018-06-09:
In the following link the author states: "D9 and D10, also labeled RX and TX on some boards, are also used for serial communications, including loading programs onto the device. I have read that if your program does not use serial communications, you can also use D9 and D10 as digital I/O. But, I have not been able to get this to work reliably."  So maybe I just got lucky at some point! But this was written for a NodeMCU board....maybe WEMOS does not have the same issue:  

[http://play.fallows.ca/wp/projects/electronics-projects/understanding-nodemcu-esp8266-12e-limitations/]

Jason2866, 2018-06-09:
D9 and D10 is used for communication with flash memory and cant be used with a ESP8266
D9 and D10 can only be uses with a ESP 8285! That is the reason (i think) for choosing this chip for
the CH4 because the get 2 more useable GPIOs.
So in short D9 and D10 cant be used with ESP 8266!!

Jason2866, 2018-06-09:
GPIO 9 and 10 are not configurable.  TX and RX are GPIO 1 and 3 and can be used with restrictions

closed 2018-06-11 by ascillato2

avrandrei, 2018-06-28:
Sorry for adding a comment after the issue is closed. I'm a bit late to the conversation. I've just ran into this same issue and wanted to clarify if I understood correctly. 

The sonoff with TX or RX pins on LOW at startup will not boot? Is there any way around this, or is this happening by design?

Frogmore42, 2018-06-28:
For sure you can't short TX to ground at startup.  The esp8266 will at some point during boot up try and drive the TX pin high (as it is an output).  In the best of cases, that would be a really bad thing.  In general, this will cause it to not boot.

The case for the RX pin is less clear, but I would avoid using it for anything if at all possible.

CRImier, 2019-04-23:
That's not the main reason, [the main reason is here](https://bbs.espressif.com/viewtopic.php?p=2834&sid=4a3d5aab499e0e33deabcff8a9a59d4f#p2834):

Espressif_Kelly: `Besides, U0TXD is also a strapping pin for chip_test_mode(1: normal mode; 0: chip_test_mode). Thus, U0TXD can't be pulled down just when the system is powered on.`

So, even if you don't have it shorted to ground and instead just pull a bit a current from it (i.e. a pulldown), it will still fail. I got burned on this a couple of times, for example, one time I was using a raw ESP-12 module with a USB-UART attached - but not plugged into the USB port, and it would result in the module inexplicably failing to boot (in a battery-powered prototype).

alfredopironti, 2019-04-24:
Hi @CRImier 
Thanks for shedding light on this. I think your explanation covers the issue perfectly. I was quite lucky as I pulled up the pin for no good reason, so it worked for my specific application!

Note taken, and will consider this for future applications.