arendst/Tasmota #2903: Controlling "Roller shutters" with electrodragon "ESP Relay Board IL (Inductive Load)" board?

nardev, 2018-06-03:
It looks to me as a perfect board for motor which is commonly found in aluminium blinds. That motor has two L phases and common N.

Each phase is actually different direction for the blind and stopping position is actually already adjusted in the motor while installing it.

**But here is the thing, two relays must not be on at the same time, so turning off for both relays before sending next command is essential.**

I believe also any other two channel **SONOFF** would be useful but again, same issue. The device must be treated as "single relay" board, where one relay is on/open and another relay is off/close blinds.

The is how it usually looks:

**The shutters**
[image]

**And usually the motor wiring is this:**
[image]

Frogmore42, 2018-06-03:
If you had used the search function for "rolling shutter", as the template tells you to do, you would have found 5 issues with that in it.  
https://github.com/arendst/Sonoff-Tasmota/issues/1600 has the answer you probably need.

nardev, 2018-06-03:
it's almost the same but not...

Would be nice to have that, clearly defined in the software.

Frogmore42, 2018-06-04:
Tasmota is a general purpose piece of software.  If you feel there could be more documentation on how to use it with rolling shutters, please add a section in the wiki for that.  That is one of the nice things about the wiki, you can use it to share your knowledge with those that come after you.  I don't have rolling shutters, but I have seen the same questions come up a few times.

Since Tasmota is general purpose, some things might require more configuration than you might like.  I think it is a decent tradeoff for something so versatile that still manages to fit in the constraints of the esp8266.

nardev, 2018-06-04:
Well, i understand it now.

I didn't check the code so far, but if there is some space for "modular" approach, i would definitely add some module/option which would be able to handle this.

It's actually very simple code case, i don't thing it could damage the "general purpose" approach which you try to maintain.

Anyway, than you.

closed 2018-06-04 by nardev

Frogmore42, 2018-06-04:
I didn't write the code.  Since you believe it is simple and important, take a look at the code, propose a change, code it, test it, and maybe Theo will make it part of the main code.

Many months ago, I did that with the BME280 code.  I fixed it for my specific usage and shared it.  Theo saw it, generalized it and now everyone benefits from it.

ascillato, 2018-06-04:
You can use setoption14 that is interlock for relays. As frogmore42 explain, in #1600 is the solution. Hope this helps.