arendst/Tasmota #2904: 2.3.0  vs  2.4.0  vs  2.4.1 (what is it?)

sparkgap, 2018-06-04:
After months of sweat and tears I finally got my Sonoff-Tasmota working with openHAB MQTT.   I decided to write myself a HOW-TO file, so I could duplicate my setup in the future.    When I got back  to this wiki  https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE   I got all confused again and could not remember what these numbers mean, 2.30, 2.40, and 2.4.1.

Are they Sonoff hardware model numbers? __________

Are they ESP8266 hardware model numbers? ___________

Are they software version numbers? ____________

What are the functions of each and advantages? _____________

How would I describe a HOW-TO step using them?  _______________

In the wiki I see that I needed to do setup the Arduino IDE according to these numbers.   I may have just tried something and stumbled into a working configuration.  I also recall seeing files with these numbers when I did a download, but not sure what I did with them.

sparkgap, 2018-06-04:
Thanks for the quick response.

Ok then it's software for the ESP8266 board and good explanations of issues with various versions.

The instructions are to un-comment 2.3.0 using PlatformIO, but I'm using Arduion IDE.  Now I recall running into this before.   I suppose I used whatever the default is for Ardiono IDE.  Are there wiki instructions for Arduino IDE?   To force it to 2.3.0?

sparkgap, 2018-06-04:
In the following.........
`Replace in Arduino IDE folder portable\packages\esp8266\hardware\esp8266\2.3.0 file boards.txt with the Tasmota Source code file arduino\version 2.3.0\boards.txt.`

I can not find the IDE folder portable file.

sparkgap, 2018-06-04:
Now I recall the confusion.... wiki not written well.  
```
Install ESP8266 board software

    Open Arduino IDE and select File - Preferences and add the following text for field Additional Boards Manager URLs: http://arduino.esp8266.com/stable/package_esp8266com_index.json and select OK.
    Open Tools - Boards ... - Boards Manager ... and scroll down and click on esp8266 by ESP8266 Community. Click the Install button to download and install the latest ESP8266 board software. Select Close.
 

```
First bullet point does not make sense.  If what is not available?

Second bullet point seems poorly written
I was looking for these file in Program Files (x86) not there, but.......
I found.... user/appdata/local/Arduino15/packages/esp8266/hardware/esp286/2.4.1 with a boards.txt file (232KB),
But..... There was no .../portable/.... folder in the path nor was there a 2.3.0 folder.
I assume the..../portable/...  in the path was a typo.  
Furthermore I assume 2.3.0 in the path was also was a typo.
Now.... I could replace the board.text  (232KB) in the 2.4.1 folder with the boards.txt (81KB) from the 2.3.0 downloaded source folder .   Is this correct?

Now back to the first bullet point, should that be done in addition to the second bullet point?

Frogmore42, 2018-06-04:
I would suggest PlatformIO.  It has a bit of a learning curve, but I find using it with VSCode to be a far superior experience than using the Arduino environment for anything that is not trivial.  Switching between 2.3.0, 2.4.0, and 2.4.1 is literally a one line change (single character if you like).
Why people think Arduino is a good development environment, I don't believe I will ever understand.  Arduino is great if you have never done any programming and you desire does not go much beyond blinking an led (the proverbial hello world of embedded programming).  Once you want to do something more complicated, the Arduino model makes it harder.

sparkgap, 2018-06-04:
I've spend a week on Atom and Platform IO and it is still over my head.  The same goes for VB editor, weeks more of learning.  I use notepad ++ and Arduino very intuitive IMHO and have used both for years.   I realize you prefer with VB and IO, but If someone could please clear up the Arduino IDE wiki (which is supported) , I'd have things working in a few minutes. 

It appears that there are one or two files that need replacing.   The first question is to clarify which or if both need replacing.   I found both files in the source code for the 2.3.0 library, I just need to know which files to replace in the Arduino installation.   Seems like an easy question for someone familiar with Arduino IDE.    That would save me weeks of work learning IO and VB and possibly avoiding creating more problems.      P.S.  Using Arduino 1.8.5

Frogmore42, 2018-06-04:
I think you are making my point.  Arduino is great for beginner projects, because it hides all the complexity from you.  Compiling Tasmota is not really for beginners, because there are issues with the underlying libraries/platform upon which Tasmota depends.  Getting these correct is easy in PlatformIO.  Doing it in Arduino is beyond the general capability of beginners.

Over the last year or two Arduino made it easier for people to extend the system, with the boards manager.  But, that is a complex subject that most people don't understand.  I don't typically use Arduino, because its editor is very poor.

I agree that PlatformIO can be intimidating.  It can do a lot and can seem complex, but you don't need to understand much of it to get it to build your firmware

arendst, 2018-06-05:
Pls do not update the wiki with your findings as for core 2.3.0 you would still need to do the changes as documented in the wiki. Core 2.3.0 did never have the correct linker file needed for Tasmota.

sparkgap, 2018-06-06:
I had no intention to update.  I thought you (as author with a password) were the only one who could make changes.    This is spooky, if anyone like me could change things.   I really don't understand how GitHub works.

As for my configuration, I have six sonoffs working perfectly with openHAB rules.  
I get this
`00:00:00 Project sonoff-0 Sonoff (Topic sonoff-4, Fallback DVES_07E8F4, GroupTopic sonoffs) Version 5.14.0-2_3_0`
every time I boot up so I assumed it took 2.3.0    
I could not make the changes as documented, for instance it says:
`Replace in Arduino IDE folder portable\packages\esp8266\hardware\esp8266\2.3.0 file boards.txt `
But there is no such directory path 2.3.0 and there is however a 2.4.2 directory in the path.  Purely following instructions, one can not get there.

arendst, 2018-06-06:
Just a heads-up: I do not want to sound rude but the whole thing about changing boards.txt is providing the correct 1M(No SPIFFS) environment you will need to perform your first OTA or web firmware upload. With your current 1M(64K SPIFFS) it will fail and you will start thinking again why doesn't it work.

Any way. If it works for you keep it that way...

Frogmore42, 2018-06-06:
The whole point of GitHub is to share knowledge.  So, yes anybody who has an account can change the wiki, but it does keep track of who does what and when.

As far as Arduino goes, it is great.  It simplifies things.  That is great if what you need is simple.  However, if what you need doesn't fit in the simple model, you have to figure out how to do what you need and it might be even harder.

The use of 2.3.0 is a good example.  In Arduino, there is no built-in support for the 1M memory model.  Back when 2.3.0 was new, not many projects needed that much memory (and it would be really hard to manage with the Arduino development environment).  There is a way to make it work, since Arduino is really just a thin layer over C++.  C++ has never been known to be easy, but most would agree it is pretty powerful.  Arduino hides much of the (usually necessary) software engineering knowledge, which makes it easier to do many basic things.

If you are not developing software, you don't need to know it.  If all you want to do is run Tasmota, you don't even need Arduino.  Just install one of the pre-built binaries.  Theo has done an excellent job of making it so you don't have to pre-configure anything.  You can just flash a device and then configure it.

Jason2866, 2018-06-06:
Just my 2 cents.... I am not a programmer (salesman for PBX and IT things)
I used Arduino for "hello world" on a LC Display. And i liked it to do cute little projects
in a easy way for non programmer. 
Always interested in Home Automation someday i found Tasmota. This was exactly what i need to
build a HA system which suits my needs...
After a hard learning curve to get basics running in OpenHab, the next big step was get a Itead device
running with it.
After reading the wiki, for me was clear Platformio is the easiest way to get Tasmota compiled.
Sounds strange but if you compare what you have to do with Arduino vs. Platformio to setup
Platformio is the winner!  Followed exactly the wiki -> Installing was smooth and no question was left open
After 1 h, i had my first self compiled Tasmota. Flashing went smooth too (i was lucky here because
i had/have a Ch340 adapter which is perfect for that)
After i understand what Theo had written (in wiki) to get a working switch in OpenHab.
-This took some time (i dont know nothing about mqtt at this time)-
And did it exactly as he described, i got a working switch via OpenHab.
I felt great at this moment like a child when getting the long awaited and wanted birthday gift...
So maybe i was simply a lucky guy to get Platforrmio running (fast).
Now i use Visual Studio Code. The reason for change. I can play with OpenHab and with Tasmota
in the same environment  😃  Installing went smooth too. I just followed the wiki again.
My conclusion dont give up to early, and do read the wiki. I can speak only for me i didnt understand
always what is written there at the first time. So i read it again...

sparkgap, 2018-06-07:
Thanks to all for the help and encouragement.   This is so embarrassing I don't ever know what SPIFFS is.   I always thought it partitioned EEPROM memory chip space into program storage area (static) and variable storage ares (dynamic), but then "no SPIFFS" would make no sense.   I always used whatever the instructions said to use, and it worked, never having to worry about what SPIFFS is.    So tonight I Google up on SPIFFS.    To be continued tomorrow.......

localhost61, 2018-06-15:
@sparkgap
IMHO Arduino is enough if you don't need to code, just edit user_config.h in notepad++ and compile.
It's default behavior that is to open at once all source files of the project folder make it very long to load.

Even if I switched to PlatformIO I can also compile Tasmota under Arduino and I even upgraded it's ESP2866 platform to the latest 2.4.1. So I can tell you that with a standard install (not portable) and under Win 8.1, Arduino IDE 1.85 put it's data in  %localappdata%\Arduino15\packages\esp8266\hardware\esp8266\

To change the platform in Arduino, open the board manager, like this :

[image]

And find the ESP8266 package, you can change it's version by a click on "More info" :
[image]

sparkgap, 2018-06-21:
I'm not really ready to close this yet.  To recap..... my objective;  Initially I was using HTML POSTS to control my sonoffs and was completely happy without Tasmota, however my c++ code would lock-up randomly every few weeks.   Cycling the power would reset everything.   The lock-up appeared to happen when the wifi signal became marginal and one of the wifi libraries would loose a hand-shake due to poor error recovery (or something) in one of the wifi libraries.    

So.... I decided to try tasmota-sonoff. hoping it would be more robust and have some sort of watch-dog-timer to reset if there was a lock-up.   When installing tasmota I chose to fall back to 2.3.0 because of some comments about wifi stability.   Unfortunately with the Arduino 2.3.0 board manager the flash size could only be set (via Tools menu pick) to 1M(64k-SPIFFS) whereas I am told that 1M(no-SPIFFS) is required if I want to do OTA upgrade.  NO, I do not want to switch to and learn platformIO as some have suggested (I tried two times and got lost in the busy GUI).    Before permanently installing too many of these sonoffs, I was wondering if I should use the latest 2.4.1.   In other words....  is there really a wifi stability (lock-up) issuie with Ardiono Board Manager 2.4.1 ??????????????????

ascillato, 2018-06-21:
2.4.1 have problems
2.3.0 is more sable

sparkgap, 2018-06-21:
ascillato, Thanks

That's my thinking based on your and others report.   So I will stick with Arduion 2.3.0 and worry about the OTA upgrade later.    So far 2.3.0 is serving me well on several sonoffs.   My old HTML code would have locked up by now....going on 3 weeks and no problems w/  2.3.0.

I'll leave this issue open for a while in case someone wants to comment on why Arduino 2.3.0 does not allow selection of 1M (no-SPIFFS)  as well as how 1M(64K-SPIFFS) will affect me when doing OTA upgrades.  Furthermore,  I really don't understand the SPIFFS thing other that 64K is set aside for something.

sparkgap, 2018-06-22:
Closing this issue and opening another with more appropriate subject line.
See.... "Can't set Arduino Flash Size to 1M(No-SPIFFS) with 2.3.0"

closed 2018-06-22 by sparkgap