arendst/Tasmota #2905: Sonoff RF Bridge pairing with 433Mhz RF single channel relay

vaidyasr, 2018-06-04:
With the Sonoff RF Bridge can I control a 433Mhz RF single channel relay without an remote?. 
The 433Mhz RF channel is [https://www.aliexpress.com/item/433MHz-RF-Single-Channel-Remote-Control-Switch-Remote-Control-Relay-Relay-Receiver-Module-433MHz-Wireless-Frequency/32826645756.html?spm=a2g0s.9042311.0.0.438d4c4dyygMtn](here).

ascillato, 2018-06-04:
Please, check in itead website the protocols that sonoff-bridge supports and ask the manufacturer of that rf relay if that protocol is compatible with theirs. Thanks.

Jason2866, 2018-06-04:
@vaidyasr 
Will work receiver is a module who is learning the codes for on and off.
So you can simply send the default codes from the rfbrige and let the receiver this codes learn.
I use a similiar module for that use case.

vaidyasr, 2018-06-05:
Thank you for the replies. From various tutorials I can see the remote can be used to do initial pairing and then RF bridge can control. But can I pair a RF relay with RF bridge directly without an remote?.

arendst, 2018-06-05:
@vaidyasr yes. By default all 16 buttons send a uunique rf code. If you have more bridges part of the unique code can change too

As always, see wiki.

vaidyasr, 2018-06-06:
Am trying to pair with the relay and got the following error.
`05:27:11 CMD: Rfkey1 2
05:27:11 RSL: Group 0, Index 1, Command RFKEY, Data 2
05:27:11 MQT: stat/sonoff_bridge/RESULT = {"RfKey1":"Start learning"}
05:27:11 SER: Received A0 55
05:28:11 SER: Received A2 55
05:28:11 MQT: stat/sonoff_bridge/RESULT = {"RfKey1":"Learn failed"}`

Jason2866, 2018-06-06:
Hello @vaidyasr 
did i understand you false? You have no remote, you said.
What should the bridge learn?? There is no signal sended.
The receiver (switching device) has to learn the sended codes from RfBrigdge!

vaidyasr, 2018-06-07:
@Jason2866, yes I don't have any remote to pair, ordered one. I thought the RFBridge can send the similar signal like RF-Remote, so that I can eliminate a purchase of remote.
If I use a remote to pair a single channel RF relay and then with RF-Bridge, an I use the same remote to pair another RF Relay?

Jason2866, 2018-06-07:
Facts :
- You need no remote to switch the device. The RF Bridge is the „Remote“ Just do the command
   rfkey1 to send Key 1 and rfkey2 to send Key 2
- You CAN use a extra remote > Put the receiver in learning mode again to learn this remote

vaidyasr, 2018-06-07:
@Jason2866 
please bear with my novice question:
I bought an single RF relay and 4 channel RF relay. I ordered an 4 button RF remote too.
So, to configure the single RF relay, if I am using one of the buttons in "4 button RF remote", can I use the same "4 button RF remote" to configure the 4 channel RF relay?. or will it conflict with single channel RF relay?. i.e. if I press button 1 on the RF remote, will it activate both RF relays?

Jason2866, 2018-06-07:
One button on the remote can only send always the same code.
So if you learn this code on more devices every device will react on this code when learnt.
So if you want to switch the devices independently you have to use different buttons on the remote!
To answer your question, yes it will conflict with a remote with only 4 buttons.

vaidyasr, 2018-06-08:
@Jason2866 
The RF bridge basically learn key press from remote like
> 15:54:41 BRG: Received A3 20 DA 01 18 03 34 2E 1A 22 55 

Can't we save/insert the above code into the RF bridge without an remote to send the above code?

Jason2866, 2018-06-08:
Yes, just take a look in the wiki RF Bridge Commands

vaidyasr, 2018-06-08:
Yeah, I am reading that and don't understand. How to send the above hexadecimal key to RF bridge

Jason2866, 2018-06-08:
Where did you got this code? Or in a other question what have you used to receive this code?
A typical recieved message from the RF Bridge looks like that
`17:24:46 MQT: tele/sonoff-82589D/RESULT = {"RfReceived":{"Sync":11170,"Low":360,"High":1090,"Data":"151454","RfKey":"None"}}`

vaidyasr, 2018-06-08:
that was a example output of some of the post listed here. I was looking to use that code to activate my RF relay. I am still waiting for the RF remote to arrive to my location, in the mean time I thought the RF bridge can be programmed with a user RFKey so that no remote is required.

Jason2866, 2018-06-08:
For programming your receiver there is no setup on the RF Bridge neccersary!
You just can press in web console the blue "1" and the bridge sends the standard "factory key"
When you receiver is in learning mode. Just simple as that.
If you have got your remote, you just need the command 
RfKey<x>  4  Save RF data using RfSync, RfLow, RfHigh and last RfCode parameters
where x is one of the 16 Buttons (Keys) that the brige can store

vaidyasr, 2018-06-08:
My doubt is here https://github.com/arendst/Sonoff-Tasmota/issues/1001 which the RF bridge learn the code by entering manually. It says implemented but don't know how to use it

Jason2866, 2018-06-08:
Wait until you receive your equipment has arrived, it is easier than it sounds....

Jason2866, 2018-06-08:
Which remote have you ordered? Nearly all cheap ones are supported by the RF Bridge (to learn).
So case #1001 will not affect you.

vaidyasr, 2018-06-08:
Initially ordered a single channel RF relay and I thought the RF bridge can control without an remote. Then I ordered a 4 channel RF relay. When the single channel RF relay reached me, I can't control via RF bridge, then I ordered a 4 channel remote https://www.aliexpress.com/item/QIACHIP-433MHz-4-Channel-Remote-Control-light-switch-Learning-Code-1527-Transmitter-Wireless-Key-Fob-for/32846567142.html 
Now the question is I cannot reuse one of the button of 4 channel remote for the single channel, do I need to order an single button remote to control the single channel RF?. Then both the remotes will be shelf, as the relays can be controlled via RF bridge?

Jason2866, 2018-06-08:
The single relay module should be able to be controlled by the rf bridge.
Just do it like i described in the post with the picture!!
You can control the single relay device with both remotes and the rf bridge.
You have to learn the transmitted codes in the relay module!
The codes from the 4 channel remote transmitter  can be perfect learned too by the rf bridge!
So you can use the 4 channel remote to control the 4 channel relay.
The one? button remote to controll the single relay device.
With the rf bridge you can controll both relay devices (one channel and 4 channel)
Everything will work, like you want to.

Jason2866, 2018-06-08:
The rf encoding "1527" is perfect for the tasmota device!