arendst/Tasmota #2906: MagicHome LED strip controller programs

powerpolly, 2018-06-04:
big thanks to tasmota for the flash of these led stripes! this firmware really enables these controllers to be integrated into my homeautomation system (openHAB). I can remotely control colors and power it on/off. I also added a conventional switch so the WAF is met (this was not possible with the original firmware). the only thing I`m missing is the selection of programs like the original firmware had. eg my fishtank has a sunrise and sunset program. my ledstrips throughout my house act as silent alarm system. they flash blue light in a sequence that simulate police lights. in tasmota firmware these programs are called schemes but how can I create custom schemes? 
thanks

ascillato, 2018-06-04:
See rules in the wiki

powerpolly, 2018-06-04:
I saw the rules section already, but I`ve no clue how to let my leds blink in blue color. any more detailed documentation available? thanks!

ascillato, 2018-06-04:
No. To make your schemes with rules you have to use your imagination.

If you want that police type you can, for example, make an event rule that
uses ruletimer to turn it on and then off the blue color.

powerpolly, 2018-06-05:
unfortunately I don`t have a developer imagination...
how to trigger a rule via MQTT?
I can`t figgure out how to switch on/off a LED for 0,5 sec...
can someone provide an example for a rule?

ascillato, 2018-06-05:
I have re read my previous comment and it "sounds" a bit rude. Sorry, I did not want to meant that. I should make a better choice of words. I was trying to say that it is possible with rules and that all the info is in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki).

About the example (make the **blueflash** event in the rules):

We want to make the ledstrip to:
- Power on and get blue and then, after 500 miliseconds, to power off
    (event#blueflash do backlog pulsetime1 5; color 0,0,255; ruletimer1 1 endon)
- Repeat the cycle every 1 second
    (on rules#timer=1 do backlog color 0,0,255; ruletimer1 1 endon on)
- Stop by command/mqtt
    (on event#stopblueflash do backlog rule 0; pulsetime1 0; ruletimer1 0; power1 0; rule 1 endon)

So, everything together:

1- Go to console and type:
```
rule 1
rule on event#blueflash do backlog pulsetime1 5; color 0,0,255; ruletimer1 1 endon on rules#timer=1 do backlog color 0,0,255; ruletimer1 1 endon on event#stopblueflash do backlog rule 0; pulsetime1 0; ruletimer1 0; power1 0; rule 1 endon
```

2- To start
by commands in console: `event blueflash`
by mqtt: `cmnd/sonoff/event blueflash`

3- To stop
by commands in console: `event stopblueflash`
by mqtt: `cmnd/sonoff/event stopblueflash`

powerpolly, 2018-06-06:
no worry Adrian, it`s my fault that I didn`t make clear that I`m a newbie to SW programming. I was looking for a kind of rule tutorial for dummys... 
I tried your rule but the only thing happens is that the magichome LED controller reboots.
so I tried to run a simple rule and it also reboots the controller, I did:
`rule 1
rule on event#green do color 00FF00 endon
event green`
any idea what could be wrong?

ascillato, 2018-06-06:
Can you see the restart reason in the information menu or in status 0 on the console?

If the reason is _Exception 28_ might be a memory problem, meaning that you should recompile tasmota with less features and upload again. This happened to me and I solve just disabling sensors I do not use.

powerpolly, 2018-06-06:
I have two identical controllers. one reboots with 
tele/LED_Bad/INFO3 = {"RestartReason":"Hardware Watchdog"}, the other one with
tele/LED_Office/INFO3 = {"RestartReason":"Software Watchdog"}
they both have the IRrecv on GPIO14, LED_Office has also a switch on GPIO1. I unassigned all of them but they still reboot with the watchdog error.

ascillato, 2018-06-06:
ok. please share the output on the console of the command `status 0`

powerpolly, 2018-06-06:
I updated LED_Bad from tasmota 5.13 to 5.14. restartreason is now also software watchdog.
`16:24:07 CMD: status 0
16:24:07 MQT: stat/LED_Bad/STATUS = {"Status":{"Module":34,"FriendlyName":["LED_Bad"],"Topic":"LED_Bad","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
16:24:07 MQT: stat/LED_Bad/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff-minimal.bin","RestartReason":"Software Watchdog","Uptime":"0T00:00:31","StartupUTC":"2018-06-06T14:23:36","Sleep":0,"BootCount":41,"SaveCount":886,"SaveAddress":"F5000"}}
16:24:07 MQT: stat/LED_Bad/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-06-05T10:46:02","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:24:07 MQT: stat/LED_Bad/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["pollyhome",""],"TelePeriod":300,"SetOption":["00808009","55818000"]}}
16:24:07 MQT: stat/LED_Bad/STATUS4 = {"StatusMEM":{"ProgramSize":528,"Free":472,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
16:24:07 MQT: stat/LED_Bad/STATUS5 = {"StatusNET":{"Hostname":"LED_Bad-6036","IPAddress":"192.168.1.143","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:C0:77:94","Webserver":2,"WifiConfig":3}}
16:24:07 MQT: stat/LED_Bad/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.151","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C07794","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:24:07 MQT: stat/LED_Bad/STATUS7 = {"StatusTIM":{"UTC":"Wed Jun 06 14:24:07 2018","Local":"Wed Jun 06 16:24:07 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"05:48","Sunset":"21:49"}}
16:24:07 MQT: stat/LED_Bad/STATUS10 = {"StatusSNS":{"Time":"2018-06-06T16:24:07"}}
16:24:07 MQT: stat/LED_Bad/STATUS11 = {"StatusSTS":{"Time":"2018-06-06T16:24:07","Uptime":"0T00:00:31","Vcc":3.106,"POWER":"OFF","Dimmer":100,"Color":"0000FF","HSBColor":"240,100,100","Channel":[0,0,100],"Scheme":0,"Fade":"ON","Speed":2,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"pollyhome","RSSI":80,"APMac":"3C:46:D8:A0:D1:28"}}}`

ascillato, 2018-06-06:
Your config seems ok to me.

did you update by OTA?

powerpolly, 2018-06-06:
first update was with serial/USB converter. then I modified the section "MagicHome" in sonoff_template.h because the PWM to GPIO assignments were wrong and showed wrong colors. then I had to OTA to sonoff_minimal because of the limited flash size and then I was able to OTA my customized build.

ascillato, 2018-06-06:
Ok, I was asking that because in just one device has happened to me that updating from an old version, some values that remains in the flash, were making conflict. I suspect something of the sdk (esplib).

In my case, I solve that erasing all the flash (esptool.py --port COM5 erase_flash) and uploading again by cable. I know. It is a pain to do that, but in my case, that solves my software watchdog reboot.

ascillato2, 2018-06-27:
Hi, Have you manage to solve your issue? If yes, please close it. Thanks :+1:

closed 2018-06-27 by ascillato2