arendst/Tasmota #2907: External Antenna

Lawrencezarb, 2018-06-04:
Has anyone managed to connect an external wifi antenna to a sonoff basic to improve Wifi signal.

I have a unit that is 10m from the router but only receives about 12% signal strength...!

wasi54, 2018-06-04:
This is often du to signal reflection which can be solved with repostioning. Try to turn the sonoff by 30, 60, 90 degre or upwards (vertical position)
and/or reduce the AP transmission power down to minimum (4 mW).

ascillato, 2018-06-04:
Hi, wasi54 is right. Try to repositioning the sonoff. There is no easy way to modify the hardware for that.

closed 2018-06-04 by ascillato

sandos, 2019-05-14:
Just a comment. I compared the RSSI of 2 Wemos D1 mini:s with the Sonoff Basic. The sonoff basic and one of the minis was ~1m from my router. The Sonoff had 20db less signal strength! Also, a mini that was about 10 m away, with a wall in between had much better RSSI than the sonoff that was very close to the router. These antennas are really crappy, the mini also uses PCB antennas. I suspect they somehow failed with these. They are maybe shielded by the groundplane in the PCB?

And yes, these antennas are very directional somehow. They seem to like it best to face with their flat downside towards the router, where the signal does not have to traverse the PCB.

GoBooth, 2023-01-28:
> Just a comment. I compared the RSSI of 2 Wemos D1 mini:s with the Sonoff Basic. The sonoff basic and one of the minis was ~1m from my router. The Sonoff had 20db less signal strength! Also, a mini [...]

sometimes when you're too close to the router, it appears like its bad signal. a few factors are in play here. is your Router transmitting 2.4ghz on HIGH? most people have their routers configured like this. This is the same as someone screaming in your face. you're very close, but it's hard to understand what they're saying. it's just noise.

barbudor, 2023-01-29:
Exactly, That's saturation which introduce distortion decreasing the signal to noise ratio.

And antennas are, IMHO, a hard science. And pcb antennas even harder.
Length, section, impedance etc there are so many parameters that it's easy to ruin the antenna with a little drift from the specs.