arendst/Tasmota #2908: "

petonic, 2018-06-04:
**Describe the bug**

A clear and concise description of what the bug is.

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues and in the wiki
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-06-04:
Error?

Closing.

closed 2018-06-04 by ascillato

petonic, 2018-06-04:
Well, there is an error, but I'm trying to determine if it's in my head or in the code :-)  If it's the latter, i'll come back with more details.  Premature.  Thx.