arendst/Tasmota #2910: Sonoff dual R2 external switches and programming mode

Kepete, 2018-06-05:
I have a Sonoff dual R2 behind my usual wall switch. Everything works fine until the power is cut and reapplied. If this happens then the connection between GND and BUTTON 0 ca not be active, otherwise it starts in programming mode. Is it possible to turn off it somehow? 
Or make it quit of programming mode automatically after some time? 

I really want to use my switches that I have, I would not wan't to change the switches.

Leseratte10, 2018-06-05:
The bootloader starts on bootup if GPIO0 is pulled low. There is nothing you can change about that. Why don't you wire your switch to another GPIO pin?

Kepete, 2018-06-05:
I wan't to control both relays from switches. As the wiki says:  

> BUTTON 0 (= GPIO0) connected to GND controls relay 1 and BUTTON 1 (= GPIO9) connected to GND controls relay 2.

Is there an other possibility to connect the switch and control relay 1 ?

Frogmore42, 2018-06-05:
Since you have it behind the wall plate, you can use any of the GPIOs for anything (within the standard constraints).  You will need to select generic device and then configure both switches and relays.  Use what the dual uses expect change the GPIO for switch 1 to the new one.

ascillato, 2018-06-12:
Hi

If your issue is solved, please close it. Thanks :+1:

BigBill2000, 2018-06-21:
I have the same problem, but I think I have found a solution. In the sonoff_template.h file, under Sonoff Dual R2 change GPIO10 button from GPIO_KEY1 to GPIO_USER. Now you can set GPIO10 for switch 1 or 2 and GPIO9 for the other switch. Now just solder a wire to the push button and you are good to go!

closed 2018-06-27 by ascillato2

mdo-git, 2020-04-23:
Have confirmed adfra's solution works perfectly.