arendst/Tasmota #2916: Switching relay with ground pulses on requested state transition

thilokru, 2018-06-05:
I've got a device, in which the relay is toggled via ground pulses on two seperate IO pins. I'm currently using a work around, where one pin is constantly low, while the other one switches the relay accordingly. However, this heats up the voltage regulator severely, which I'd like to prevent.

If possible, implementig a pin type for generic devices, which sends a ground pulse on the pin when the relay state should transation from off to on or vice versa, would solve this problem. Additionally, for future devices, an inverted mode should be considered.

I've considered continuing to use the workaround described above, but due to the heat production and the environment in which this device is used, I do not consider this save.

I'd like to give the name of the device, but it is a no-name device. [See this link.](https://www.obi.de/hausfunksteuerung/wifi-stecker-schuko/p/2291706)

arendst, 2018-06-05:
See #1988

closed 2018-06-05 by arendst

ascillato, 2018-06-05:
Hi

If it is heating, put a resistor on that pin to ground to limit the current. 

Inversed mode is implemented. See relay1i options.

dherrendoerfer, 2018-06-07:
I think this issue is valid.
I have the same plug, and instead of a latching relay it has a set/reset flipflop connected to I/O5 and I/O12.
The maker of this thing used this to make the device reboot-proof (cool idea, I believe). So if the 
ESP resets for any reason the state of the relay is not affected and can be safely set after the
restart of the controller.