arendst/Tasmota #2918: Only 1 gpio works with my IR receiver on wemos d1 mini

gymolnargabor, 2018-06-05:
Hi all,

I have problems with connecting more than 1 sensors to my Wemos D1 mini.
I have tried with several sensors, first with an IR reciever. I have configured all the GPIOs to IRrecv, but the device only works with the GPIO2, the others did not receive any signals (same receiver, I have just moved the out pin). Then I took my BH1750 sensor, I did the same thing, I have set all the GPIOs to SCL and SDA (always next to eash other), and it was working only on the GPIO2-GPIO0 combo.
I have flashed the wemos with the newest devlopment firmware over arduino IDE, and I have only changed the wifi settings and enabled the analog input. And I changed to 18 Generic module type.
What did I do wrong? How can I use this 2 sensors together?

arendst, 2018-06-05:
If you know that only one of each type is supported AND that the software scans the gpios from 0 up you then understand why only the first detected gpios are found and used.

Solution: make your mind up and select only 1 gpio fir irreceive and select 2 other gpios for scl/sda.

gymolnargabor, 2018-06-05:
ok, i did not know that.
thank you!

closed 2018-06-08 by ascillato2