arendst/Tasmota #2919: SK6812 Doesn't Seem to be Working Properly

MrPaulAR, 2018-06-06:
Issue:
The LED strip isn't lighting completely.  It dies after about 30 pixels.

Tasmota Config
Module: Sonoff Basic
GPIO14: WS2812
Pixels: 600 (set via console)
Color: #RRGGBBWW (set via console based on commands doc)

LED strip is powered via 5v 30amp on just one end.  This will limit powering of the full strip but I would still expect to see more pixels lit and the ones that are lit should be acting as instructed.

The LED strip is behaving very oddly.  Only the first 60ish pixels are lighting even though I changed the count to 300 (I didn't count the number illumined).  When I run different schemes the colors don't change.

Another example is if enter color #ff00000000 I get the following output on the console but the leds that are lit change to a static rainbow color.

```
02:13:46 CMD: color #ff00000000
02:13:46 RSL: stat/sonoff/RESULT = {"POWER2":"ON","Dimmer":100,"Color":"FF0000","HSBColor":"0,100,100","Channel":[100,0,0]}
```

Frogmore42, 2018-06-06:
You might want to look here
https://github.com/arendst/Sonoff-Tasmota/issues/2357

How are you driving the strip?
Level converter or directly?

DMA or not?

MrPaulAR, 2018-06-06:
I'm using a precompiled version of tasmota (5.14.0) and don't see an option to change between DMA and BitBang

Frogmore42, 2018-06-06:
It's a compile time option.  The default is not DMA.  If you have the ability to compile it, you could try the DMA option to see if it works better.

I don't use led strips so I don't have any other insights.

Jason2866, 2018-06-06:
I only have a ws2812 24 pixel ring. But this works as it should (clock)
Do you have set really the correct pixel amount?
You need a decent extern power supply for 600 pixels. What are you using?
One end powering is a problem. To many for that.

closed 2018-08-05 by stale[bot]