arendst/Tasmota #2920: Sensor name ds18x20 rule

1igeza, 2018-06-06:
Hi,
I'm trying to use rules to send temperature to domoticz. I'm using #define USE_DS18x20_LEGACY. 
I tried different device name (DS2, DS18B20-2, DS18B20-DS2) and can't make it work. 
Could someone  check what I'm doing wrong?

06:29:11 MQT: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","Free":412,"Rules":"on tele-DS18B20-2#Temperature do publish domoticz/in {"idx":54,"nvalue":0,"svalue":"%value%"} endon"} 

06:28:16 MQT: tele/sonoff/SENSOR = {"Time":"2018-06-06T06:28:16","DS18x20":{"DS1":{"Type":"DS18B20","Address":"28FFE9A9C016042F","Temperature":21.1},"DS2":{"Type":"DS18B20","Address":"28FFADAAC317052F","Temperature":22.4},"DS3":{"Type":"DS18B20","Address":"28FF1FD2C3170557","Temperature":35.4}},"TempUnit":"C"}

Main webpage:

DS18B20-1 Temperatura	21,1°C
DS18B20-2 Temperatura	22,4°C
DS18B20-3 Temperatura	35,4°C

Sonoff-Tasmota 6.0.0a by Theo Arends

mikeo999, 2018-06-07:
Tasmota send only 1 temperature to Domoticz as far as I know.

Maybe you can try this. 
https://github.com/arendst/Sonoff-Tasmota/issues/860

Only problem is that after every update you need to change the code again. So i hope this will be change in the source.

mikeo999, 2018-06-07:
Nice did not know that. Now trying to understand it. 
I don't see how you can fill the right IDX's from Domoticz with it. I only see one IDX number 54 but how to tell what the other IDX numbers for the other tempsensors in Domoticz are.

arendst, 2018-06-07:
Two items here.

I think the message as received in the initial issue will not work with rules currently. This message is send using ds18x20 legacy.

You will need to select the other ds18x20 driver to get a message that will work with rules.

It then becomes possible to use rules like:
```
on tele-DS18B20-1#Temperature do publish domoticz/in {"idx":54,"nvalue":0,"svalue":"%value%"} endon
on tele-DS18B20-2#Temperature do publish domoticz/in {"idx":57,"nvalue":0,"svalue":"%value%"} endon
on tele-DS18B20-3#Temperature do publish domoticz/in {"idx":58,"nvalue":0,"svalue":"%value%"} endon
```
so for every sensor you'll need to define an additional rule.

1igeza, 2018-06-07:
With the other driver it's working correctly. 

```
14:21:32 MQT: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","Free":312,"Rules":"on tele-DS18B20-2#Temperature do publish domoticz/in {"idx":54,"nvalue":0,"svalue":"%value%"} endon on tele-DS18B20-3#Temperature do publish domoticz/in {"idx":55,"nvalue":0,"svalue":"%value%"} endon"}
14:21:43 MQT: domoticz/in = {"idx":6,"nvalue":0,"svalue":"30.1","Battery":60,"RSSI":6}
14:21:43 MQT: tele/sonoff/SENSOR = {"Time":"2018-06-07T14:21:43","DS18B20-1":{"Id":"0416C0A9E9FF","Temperature":30.1},"DS18B20-2":{"Id":"0517C3AAADFF","Temperature":49.1},"DS18B20-3":{"Id":"0517C3D21FFF","Temperature":73.1},"TempUnit":"C"}
14:21:43 RUL: TELE-DS18B20-2#TEMPERATURE performs "publish domoticz/in {"idx":54,"nvalue":0,"svalue":"49.1"}"
14:21:43 MQT: domoticz/in = {"idx":54,"nvalue":0,"svalue":"49.1"}
14:21:43 RUL: TELE-DS18B20-3#TEMPERATURE performs "publish domoticz/in {"idx":55,"nvalue":0,"svalue":"73.1"}"
14:21:43 MQT: domoticz/in = {"idx":55,"nvalue":0,"svalue":"73.1"}
```

closed 2018-06-07 by 1igeza

Beef-well, 2018-10-30:
Does this mean that the Legacy_DS18B20 drivers will not work at all with DisplayText?