arendst/Tasmota #2921: Feature Request - RF Bridge - Support Status pings sent from Sonoffs PIR

PrimusNZ, 2018-06-06:
According to the leaflet bundled with a Sonoff PIR, The PIR sensors will send an RF message periodically with information such as battery level.

I've been monitoring MQTT and the Tasmota console, and I don't see these status pings received by the RF Bridge.

If Tasmota doesn't already support receiving these, could it be added. Or if it's already there, how does one expose these pings to MQTT.
I want to know when the batteries are getting low in the PIR sensors and Door Sensors (Reed switches)

arendst, 2018-06-06:
Well if it doesn't receive a ping it won't show it either. You probably have to let the bridge learn these ping codes before it can act on it.

For now enable more logging (weblog 3) to see if any unknown code is received.

andreas6, 2018-06-06:
I am very interesting about this feature. Please report back if you figure it out.

ascillato, 2018-06-06:
Hi,

If your PIR has that option, you will have inside the PIR a jumper to change to config or test mode (depending on the brand). In that mode will send the RF commands (battery, etc), so your sonoff RF Bridge can learn them. For more information you have to see your RF PIR manual.

Tasmota already support receiving these RF codes. There is nothing to implement. Just need pairing configuration as the PIR manual states.

ErikAndren, 2018-06-06:
Has anyone documented the PIR messages?

ascillato, 2018-06-06:
The PIR sends its own codes. Each PIR will have different codes. Like the buttons of a RF switch. each button on each controller will be different

ErikAndren, 2018-06-06:
Yes, but is it just an unique hash or does it contain information like a timestamp serial no etc.

ascillato, 2018-06-06:
unique code for each feature. One unique code for detection, one unique code por low battery, one unique code for tamper (open case). exactly as 4 buttons of a RF switch

ErikAndren, 2018-06-06:
Ok, is this documented somewhere?

ascillato, 2018-06-06:
Yes on the manual of the PIR.

ascillato, 2018-06-06:
For the original question of this issue:

[image]

Here in the manual you can see the config jumpers.

There you can select which mode you want.

ascillato, 2018-06-06:
Ah, another thing.

The box of the reed switch from sonoff say that they inform battery level. Well. It does not. It does not have the circuit for that.

The PIR inform that correctly. The reed dont. The reed only sents a code when it opens only. Just that. The closing or batery is not informed.

ascillato, 2018-06-06:
@PrimusNZ 

If your question is solved, please close this issue. Thanks :+1:

PrimusNZ, 2018-06-06:
> unique code for each feature. One unique code for detection, one unique code por low battery, one unique code for tamper (open case). exactly as 4 buttons of a RF switch

I believe this is incorrect. My PIR (Same as the one you have pictured) has been learnt, and I see the same code for all functions and no code for status information.

> 10:28:19 SER: Received A4 30 B6 01 9A 04 C4 D3 26 BE 55
10:28:19 MQT: tele/rfbridge/RESULT = {"RfReceived":{"Sync":12470,"Low":410,"High":1220,"Data":"D326BE","RfKey":1}}

This is the ONLY message I receive, in any of the modes (Test, Normal and Code learning mode)
It took me an age to work out how to learn the code in the first place because the docs on the RF Bridge page do not specify how to pair a device with Tasmota. - Eventually learnt using `RfKey1 2`

ascillato, 2018-06-07:
Oh

Your are right! I tested the one of the picture and you are right! It is the same code for everything!

So I was confusing this one with another that I have installed that have all those features. 

Sorry my mistake.

So, summarizing:

* The PIR from Sonoff only sends one code for motion and the same for the tamper. It does not send any status or battery. The manual lies.
* The Reed Switch from Sonoff only sends one code when opening. Not when closing. The manual lies.

PrimusNZ, 2018-06-07:
Okay, Nice to know im not doing something stupid - Serves me right for following the manual!

No worries, I'll close this issue - Might have to have a hunt for a device that does send battery information cause I'd like to know when they're dying

closed 2018-06-07 by PrimusNZ

ascillato, 2018-06-07:
@PrimusNZ , Thanks for pointing this now. I was convinced that those PIRs were complete. I was going to buy a batch. Thanks, now you save me some wasted money. :+1:

Zoltant7, 2020-01-22:
> I believe this is incorrect. My PIR (Same as the one you have pictured) has been learnt, and I see the same code for all functions and no code for status information.
>
>
> This is the ONLY message I [...]

What do you mean that the PIR2 code was learned, with the  Rf bridge you can listen to unlimited codes,no need to pair or anything, only if you want to send  code with the Rf Bridge. Am I missing something, please correct me....

My setup was working OK for 2 Years until now i getting some unknown codes,I'm using the Tasmot 8.1 firmware with No Portisch flashed!

The problem: i use to receive the The EE66EE ,which is the correct PIR2 code for me(and i got only 1 RF sender), and i got these now in one hour,i have the default PIR2 jumpers in place,no neighbor with 433 Mhz sender
Payload: FFFFFF, with value template Template("{{value_json.RfReceived.Data}}")
Payload: 773377, with value template Template("{{value_json.RfReceived.Data}}")
Payload: F08B01, with value template Template("{{value_json.RfReceived.Data}}")
Payload: FFFFFF, with value template Template("{{value_json.RfReceived.Data}}")
Payload: 773377, with value template Template("{{value_json.RfReceived.Data}}")
Payload: 773377, with value template Template("{{value_json.RfReceived.Data}}")