arendst/Tasmota #2922: MQTT CLIENT

tcpipchip, 2018-06-06:
Sir, is it possible the Sonoff-Tasmota be a MQTT client, and not a MQTT server ?

Then i want to publish/subscribe data on MQTT cloud!

I want to send commands to SONOFF using the MQTT cloud and read data!

Miguel

Leseratte10, 2018-06-06:
Tasmota is a MQTT client. Just enter server name, port and credentials in the settings.

tcpipchip, 2018-06-06:
Thanks for the fast answer!

Do you mean here ?

// -- MQTT ----------------------------------------
#define MQTT_USE               1                 // [SetOption3] Select default MQTT use (0 = Off, 1 = On)

#define MQTT_HOST              "m14.cloudmqtt.com"                // [MqttHost]
#define MQTT_FINGERPRINT1      "A5 02 FF 13 99 9F 8B 39 8E F1 83 4F 11 23 65 0B 32 36 FC 07"  // [MqttFingerprint1]
#define MQTT_FINGERPRINT2      "A5 02 FF 13 99 9F 8B 39 8E F1 83 4F 11 23 65 0B 32 36 FC 07"  // [MqttFingerprint2]
#define MQTT_PORT              1883              // [MqttPort] MQTT port (10123 on CloudMQTT)
#define MQTT_USER              "DVES_USER"       // [MqttUser] MQTT user
#define MQTT_PASS              "DVES_PASS"       // [MqttPassword] MQTT password

closed 2018-06-08 by ascillato2

tcpipchip, 2018-07-06:
Hi team

Does TASMOTA work with MQTT that needs authentication ? (login/pass)

Leseratte10, 2018-07-06:
Yes, it does. There are two input boxes for "user" and "password".

tcpipchip, 2018-07-11:
And is it possible to make a ACCESS POINT, and not STATION ?

For example, if i press a button for a while, enter in ACCESS POINT mode to make my configs!

Leseratte10, 2018-07-11:
Yes: https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

tcpipchip, 2018-07-11:
Nice @Leseratte10 

Thanks to fast feedback! Only 57 seconds!

tcpipchip, 2018-07-11:
I see that in 
https://github.com/altelch/SonoffIR
that i can use SONOFF to transmit a INFRARED command to AIR COOLER (i am already have the sequency pulse/delay code). Where can i store this sequency to reproduce with RremoteESP8266 library  ? Or can i store on Cloud and get it ?

tcpipchip, 2018-07-11:
my air cooler are KOMECO e GREE!