arendst/Tasmota #2925: stale bot spam

davidelang, 2018-06-06:
**Describe the bug**

the stale bot seems to be repeatedly sending notices on the same issues, saying that it is closing issues, but not actually closing them

Please either turn of the stale bot or make it actually do what it's claiming to do rather than repeatedly spamming waters with the same messages

ascillato, 2018-06-06:
Hi David

Yes, was my mistake yesterday with a unintended merge. Sorry. It is Fixed now. You can close the issue if you see that it is ok now.

Thanks

davidelang, 2018-06-06:
it hasn't been a consistant issue, things seem fine for a few weeks and then I 
get hundreds of stale bot messages in a day.

If you (or someone else more active in the project than I currently am) are 
watching the repo and don't have these auto-forwarded to a folder (so that you 
will _know_ if it runs out of control), go ahead and close this issue.

David Lang

closed 2018-06-06 by ascillato2