arendst/Tasmota #2926: Sonoff SV Door Entry

echo4019, 2018-06-06:
Hi, 

I'm looking to find out if its possible to use a Sonoff SV to activate the lock release on my flat buzzer.

The buzzer I have is the Bell 801. Here is the wiring diagram - http://www.bellsystem.co.uk/manuals/801%20Cabling%20for%20DC%20Systems%20upto%2010%20way%20(PD-084%20Iss%201).pdf

I would also like the Sonoff SV to detect the when the buzzer is pressed allowing me to send an alert to my phone.

Does anyone know if this can be done?

Any help is appreciated. 

Thanks.

Frogmore42, 2018-06-07:
Just about anything is possible, but it doesn't look like it would be easy to do.

andrethomas, 2018-06-07:
Placing a normally open relay between the two red dots should work.

Turning the relay ON (i.e. closing the circuit) should cause the lock to release in the same way it would for groundsmen using the optional trades button as indicated in the circuit but you're needing to bypass the timer anyway so it does not matter.

To detect a button from the outside you'd need to check for voltage on the R line coming into the Speech Unit - It will most likely be 12V so you'd need to drop that to 3.3V to get a digital level IO signal into one of the inputs on the ESP8266 - This can be done using an LDO or if you like old school resistor dividers.

[image]

andrethomas, 2018-06-07:
There was an error in the first upload - make sure it looks like this to be correct

[image]

andrethomas, 2018-06-07:
You would essentially be doing this

[image]

echo4019, 2018-06-07:
Sorry I should have made clear in the original post that it is a shared system and I only have access to the handset in my flat.

I know that Z  shorts  to  O  when  the  lock  button  is  pressed so I was thinking I could use the Sonoff as a dry contact to imitate the button press?

Also I'm not sure how to power the sonoff. I know 12v is sent to the handset when the call button is pressed but don't know if there is any permanent voltage to it.

andrethomas, 2018-06-07:
Oh, ok.

See if there's a voltage difference between any of the wires - most likely Z or O will be ground or +12V depending on whether it pulls down or pulls up to send the signal.

It is not clear if there is other power coming to that side, but there should be (at least during the time of operation)

I would imagine that if there is not you'd need to externally power the ESP8266 - Usually, I just use an old 5V USB wall charger and a 3.3V regulator like an LD1113 to bring either the 12V you find in the device or an externally sourced 5V down to 3.3V.

echo4019, 2018-06-07:
So I've just tested the contacts with my meter and I'm getting a permanent 3.8V across R & O. R being for the microphone and O being common. Would it be possible to use a step-up converter to boost the voltage to the required 5V for the Sonoff SV?

andrethomas, 2018-06-07:
No harm in trying I guess although I suspect it would most likely introduce noise into the audio circuit.

I'd recommend using an external power source to complete your tests first to make a proof of concept and then think about using the phantom power coming over the audio line.

echo4019, 2018-06-07:
Yeah I recon an external power source is the way to go. 

Touching back on converting  the 12v call signal to 3.3v, you mentioned a LDO, do you have a link to a suitable one? Also I'm assuming this will need to connect to GPIO14 on the Sonoff to detect the change in state?

Thanks for your help

andrethomas, 2018-06-07:
LDO = Low Drop Out, but any 3.3V voltage regulator will do just fine depending on how energy saving conscious you are. For initial testing, the LD1117-3.3 is usually quite easy to obtain.

https://www.sparkfun.com/datasheets/Components/LD1117V33.pdf

They should be available from your local electronics store.

For sensing the button first determine the voltage you're needing to sense then either use a voltage dividing resistor pair or just run it through a separate LD1117 to bring it to a logic level of 3.3V

Lets say you're measuring a 12V switching voltage on the button line then do this, for example:

[image]

closed 2018-08-06 by stale[bot]