arendst/Tasmota #2927: Sonoff-RF-Bridge-433 with RF-Bridge-EFM8BB1

fpalinf, 2018-06-07:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used : Sonoff-RF-Bridge-433 
- [ x] Provide the output of command ``status 0`` :
```
02:57:12 CMD: status 0
02:57:12 RSL: stat/sonoff/STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
02:57:12 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:09","StartupUTC":"2018-06-07T01:50:03","Sleep":0,"BootCount":13,"SaveCount":27,"SaveAddress":"F9000"}}
02:57:12 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-07T03:29:13","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
02:57:12 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ztech",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
02:57:12 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":536,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","17B6179A","00000000"]}}
02:57:12 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4340","IPAddress":"10.22.108.244","Gateway":"10.22.1.1","Subnetmask":"255.255.0.0","DNSServer":"10.22.1.1","Mac":"DC:4F:22:CA:D0:F4","Webserver":2,"WifiConfig":2}}
02:57:12 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CAD0F4","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
02:57:12 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jun 07 01:57:12 2018","Local":"Thu Jun 07 02:57:12 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:47","Sunset":"20:50"}}
02:57:12 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-06-07T02:57:12"}}
02:57:12 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-06-07T02:57:12","Uptime":"0T00:07:09","Vcc":3.186,"Wifi":{"AP":1,"SSId":"ztech","RSSI":100,"APMac":"82:2A:A8:C4:70:B8"}}}

```

I did the hardware modification.

I receive correct learned sequences. Register with "rfkey1 2"
`02:57:16 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":12440,"Low":350,"High":1160,"Data":"D30DBE","RfKey":1}}
`

To be sure, the hex of EFM8BB1 is uploaded via Main menu -> Firmware upgrade -> Upgrade by file upload or is there something I missed ?
How can I be sure the EFM8BB1 is correctly flashed ?

Regards,

arendst, 2018-06-07:
Execute command rfraw aaff55 and if returns aa0055 the update went well. The original firware will not respond.

aenertia, 2018-06-08:
Does the recently added c2 flasher support allow flashing from the rfbridge itself - or will I need to run it from another esp8266 connected to the c2/k pins on the rf-bridge?

ErikAndren, 2018-06-25:
@aenertia You will be able to flash from the rfbridge itself using the firmware upgrade web interface

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks.

aenertia, 2018-08-19:
Can I reopen this - I am still unable to confirm rf update with commit 6d1ce5bd694d0546c3039744fd508b7e9448b80c (HEAD -> rf, origin/development, origin/HEAD, development)

> 

1:20:51 RSL: RESULT = {"RfRaw":"ON"}
01:21:15 UPL: File RF_Bridge_00.hex ...
01:21:23 UPL: File RF_Bridge_00.hex ...
01:22:45 UPL: File RF_Bridge_00.hex ...
01:23:13 RSL: STATE = {"Time":"2018-08-20T01:23:13","Uptime":"0T00:05:40","Vcc":3.181,"Wifi":{"AP":1,"SSId":"kainga-atawhai","RSSI":100,"APMac":"D4:6E:0E:65:A5:7D"}}
1:23:35 CMD: rfraw aaff55
01:23:35 RSL: RESULT = {"RfRaw":"ON"}
01:24:55 CMD: status 0
01:24:55 RSL: STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
01:24:55 RSL: STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:22","StartupUTC":"2018-08-19T12:17:33","Sleep":0,"BootCount":20,"SaveCount":56,"SaveAddress":"FB000"}}
01:24:55 RSL: STATUS2 = {"StatusFWR":{"Version":"6.1.1.7","BuildDateTime":"2018-08-19T23:33:53","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
01:24:55 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"172.16.254.254","LogPort":514,"SSId":["kainga-atawhai",""],"TelePeriod":300,"SetOption":["00008001","55818000","00000000"]}}
01:24:55 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":490,"Free":512,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FAFC68C","0C000000","3702179E","00000000"]}}
01:24:55 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-5884","IPAddress":"172.16.253.114","Gateway":"172.16.253.254","Subnetmask":"255.255.255.0","DNSServer":"172.16.253.254","Mac":"DC:4F:22:CB:36:FC","Webserver":2,"WifiConfig":5}}
01:24:55 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Aug 19 12:24:55 2018","Local":"Mon Aug 20 01:24:55 2018","StartDST":"Sun Apr 01 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"08:03","Sunset":"18:44"}}
01:24:55 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-08-20T01:24:55"}}
01:24:55 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-08-20T01:24:55","Uptime":"0T00:07:22","Vcc":3.180,"Wifi":{"AP":1,"SSId":"kainga-atawhai","RSSI":100,"APMac":"D4:6E:0E:65:A5:7D"}}}
--

My board looks a little different than the R2's in the Wiki -

I either get RF cannot be init, write firmware failed to RF chip; sometimes the timeout seems much longer. 

Have tried both the Power by Pin Header and jumper free pin headers + usb; with same result. 

My selector switch is in the On position (this is kinda implied in the wiki; may be useful to explicitly say you need it in the on position)

ascillato, 2018-08-19:
Reopened as requested

ascillato, 2018-08-19:
Hi, can you check if you have seriallog 0 and in the module configuration in the web menu you do not have any pin configured as software_serial?

aenertia, 2018-08-19:
I got it to flash succesfuly eventually; needed to be USB powered. I had missed the cut traces and thought it was just the jumpers that needed adding. It seems that serial power on the R2 Rev 1.0's isn't an option https://github.com/arendst/Sonoff-Tasmota/issues/3061

ascillato, 2018-08-20:
ok, great! :+1:

closed 2018-08-20 by ascillato2