arendst/Tasmota #2928: Request for adding SDM230

sensorthai, 2018-06-07:
Hi, it is possible to add new Register for SDM 230 power meter

http://www.eastrongroup.com/data/uploads/Eastron_SDM230-Modbus_user_manual_V1_4_2015.pdf
http://www.flanesi.it/blog/download/sdm230modbus/SDM230-Modbus%20protocol%20V1.2.pdf

//SDM 230 registers
#define SDM230_VOLTAGE                     0x0000                              //V
#define SDM230_CURRENT                     0x0006                              //A
#define SDM230_POWER                       0x000C                              //W
#define SDM230_ACTIVE_APPARENT_POWER       0x0012                              //VA
#define SDM230_REACTIVE_APPARENT_POWER     0x0018                              //VAR
#define SDM230_POWER_FACTOR                0x001E                              //
#define SDM230_PHASE_ANGLE                 0x0024                              //DEGREE
#define SDM230_FREQUENCY                   0x0046                              //Hz
#define SDM230_IMPORT_ACTIVE_ENERGY        0x0048                              //Wh
#define SDM230_EXPORT_ACTIVE_ENERGY        0x004A                              //Wh
#define SDM230_IMPORT_REACTIVE_ENERGY      0x004C                              //VARh
#define SDM230_EXPORT_REACTIVE_ENERGY      0x004E                              //VARh
#define SDM230_TOTAL_SYSTEM_POWER_DEMAND	0x0054		// W 
#define SDM230_MAXIMUM_SYSTEM_POWER_DEMAND	0x0056 		//W 
#define SDM230_CURRENT_POSITIVE_POWER_DEMAND	0x0058 		//W 
#define SDM230_MAXIMUM_POSITIVE_POWER_DEMAND	0x005A	 	//W 
#define SDM230_CURRENT_REVERSE_POWER_DEMAND	0x005C 		//W 
#define SDM230_MAXIMUM_REVERSE_POWER_DEMAND	0x005E 		//W 
#define SDM230_CURRENT_DEMAND	0x0102		//Amps 
#define SDM230_MAXIMUM_CURRENT_DEMAND	0x0108 		//Amps 
#define SDM230_TOTAL_ACTIVE_ENERGY	0x0156 		//kwh 
#define SDM230_TOTAL_REACTIVE_ENERGY	0x0158	 //kvarh 
#define SDM230_CURRENT_RESETTABLE_TOTAL_ACTIVE_ENERGY	0x0180                              //Wh
#define SDM230_CURRENT_RESETTABLE_TOTAL_REACTIVE_ENERGY	0x0182                              //VARh

Jason2866, 2018-06-07:
May you ask @gtortone see #2735

gtortone, 2018-06-07:
Hi @sensorthai,
you can use SDM120 instead that covers some metrics you reported...
take into account that is not possible to integrate a lot of metrics due to MQTT message size limitation...

ascillato, 2018-06-10:
Hi @sensorthai , If your issue was solved, please close it. Thanks :+1:

closed 2018-06-11 by sensorthai