arendst/Tasmota #2930: Undocumented return code MQT rc 8

mrohner, 2018-06-07:
I'm using Tasmota 5.13.1 und I'm getting this message from time to time.  I could not find any reference to this error code 8.

13:27:03 MQT: Connect failed to 192.168.178.59:8883, rc 8. Retry in 10 sec

arendst, 2018-06-07:
Good question. No idea. Are you using TLS? It's useful to provide the output of status 0 as requested when you opened this issue.

mrohner, 2018-06-07:
No, I do not use TLS
14:42:36 CMD: status 0
14:42:36 MQT: stat/sonoff21/STATUS = {"Status":{"Module":1,"FriendlyName":["BU"],"Topic":"sonoff21","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
14:42:36 MQT: stat/sonoff21/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T02:11:19","StartupUTC":"2018-06-07T10:31:17","Sleep":0,"BootCount":4,"SaveCount":19,"SaveAddress":"F6000"}}
14:42:36 MQT: stat/sonoff21/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-06-06T17:40:14","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
14:42:36 MQT: stat/sonoff21/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"192.168.178.59","LogPort":514,"SSId":["mmwireless",""],"TelePeriod":300,"SetOption":["0000200B","50018000"]}}
14:42:36 MQT: stat/sonoff21/STATUS4 = {"StatusMEM":{"ProgramSize":544,"Free":456,"Heap":10,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
14:42:36 MQT: stat/sonoff21/STATUS5 = {"StatusNET":{"Hostname":"sonoff21-BU","IPAddress":"192.168.178.21","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"5C:CF:7F:3C:14:47","Webserver":2,"WifiConfig":3}}
14:42:36 MQT: stat/sonoff21/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.59","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3C1447","MqttUser":"mqtt","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:42:36 MQT: stat/sonoff21/STATUS7 = {"StatusTIM":{"UTC":"Thu Jun 07 12:42:36 2018","Local":"Thu Jun 07 14:42:36 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"05:29","Sunset":"21:18"}}
14:42:36 MQT: stat/sonoff21/STATUS10 = {"StatusSNS":{"Time":"2018-06-07T14:42:36","AM2301":{"Temperature":27.1,"Humidity":51.0},"TempUnit":"C"}}
14:42:36 MQT: stat/sonoff21/STATUS11 = {"StatusSTS":{"Time":"2018-06-07T14:42:36","Uptime":"0T02:11:19","Vcc":3.209,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mmwireless","RSSI":82,"APMac":"24:65:11:F8:20:4B"}}}

arendst, 2018-06-07:
Ah, you're using TasmotaMqtt. In that case the RC numbering means:
```
typedef enum {
  WIFI_INIT,
  WIFI_CONNECTING,
  WIFI_CONNECTING_ERROR,
  WIFI_CONNECTED,
  DNS_RESOLVE,
  TCP_DISCONNECTING,
  TCP_DISCONNECTED,
  TCP_RECONNECT_DISCONNECTING,
  TCP_RECONNECT_REQ,
  TCP_RECONNECT,
  TCP_CONNECTING,
  TCP_CONNECTING_ERROR,
  TCP_CONNECTED,
  MQTT_CONNECT_SEND,
  MQTT_CONNECT_SENDING,
  MQTT_SUBSCIBE_SEND,
  MQTT_SUBSCIBE_SENDING,
  MQTT_DATA,
  MQTT_KEEPALIVE_SEND,
  MQTT_PUBLISH_RECV,
  MQTT_PUBLISHING,
  MQTT_DELETING,
  MQTT_DELETED,
} tConnState;
```
So a TCP reconnect (for whatever reason) but apparently you might be using the same Client Id for different devices...

closed 2018-06-07 by mrohner