arendst/Tasmota #2931: 500°F  temperature probe

alesoblak, 2018-06-07:
Hi,

is there a way to connect temperature probe/sensor on Sonoff TH that can stand temperatures up to 500°F? I would really like this for BBQ as this would enable me to see temperatures in my Kamado grill even when not at home. It is OK even if it would work with stock firmware. I dont know what probes would be ok if any.

Tnx ;)

andrethomas, 2018-06-07:
You would need a thermistor or some other similar form of a thermocouple to measure high temperatures such as those inside a BBQ

They generally require a simple resistor network connected to an ADC input so theoretically possible but not directly implementable since the currently supported sensors only go to +125 C (DS18B20 or LM75AD) with a max of 80 Degree C on the other supported sensors.

Frogmore42, 2018-06-08:
Generally, one would use a thermocouple to measure temperatures that high.  Tasmota does not support them natively.  But, there are K-type amplifiers that could be read with an adc either the internal one or an external ADS1115.  There are also K-type to digital converters, but I don't believe any are currently supported by Tasmota.

ascillato, 2018-06-12:
Hi,

If your issue is solved, please close it. Thanks :+1:

alesoblak, 2018-06-13:
I didnt actualy

closed 2018-06-13 by alesoblak

andrethomas, 2018-06-13:
@alesoblak 

Get one of these from somewhere
https://www.banggood.com/MAX31855-K-Type-Thermocouple-Thermocouple-Sensor-For-Arduino-UNO-Mega-p-1175433.html

That particular one has a SPI interface so should be pretty simple to add a SPI driver to Tasmota's code to read its values.

I think in a roundabout way your issue was solved by the previous posts, perhaps not in the direct way as you would have expected.

Please close the post unless you have further questions :)

andrethomas, 2018-06-13:
@alesoblak 
The branch at https://github.com/arendst/Sonoff-Tasmota/tree/development_display seems to support the max31855 sensor - might be useful to have a localised display anyway.