arendst/Tasmota #2933: S31 SONOFF

tcpipchip, 2018-06-07:
Sir,

Where do i define in the ARDUINO SONOFF SOURCE to use S31 sonoff ?

Today is working very nice with POWER ON and OFF

Thank you!

ascillato, 2018-06-07:
Hi,

Please, on the webmenu go to **configuration** then **module configuration** and select your **module** there.

If what you want is to define the module **before compiling and flashing**, in the _user_config.h_ file change the line 51:

from

`#define MODULE                 SONOFF_BASIC`

to

`#define MODULE                 SONOFF_S31`

If this solves your question, please close this issue. Thanks :+1:

tcpipchip, 2018-06-07:
Sir ascillato, thank you again for you fast answer!

I will test soon!!!!

Thanks

Miguel

tcpipchip, 2018-06-08:
Solved! Excelent! Thanks!

closed 2018-06-08 by tcpipchip