arendst/Tasmota #2938: Sonoff POW vs POW R2

vaidyasr, 2018-06-08:
Sonoff website claims POW R2 shows live Current (A) and Voltage (V), where as the old POW does not. But I have seen some of them posted POW MQTT values with Voltage & Current measurement. My question is, can I go ahead and buy POW which is already giving Voltage & Current values via Tasmota?

Frogmore42, 2018-06-08:
I don't have a POW R2, but I have a POW and a S31 (which uses the same power measurement chip as the POW R2).  They both measure current and voltage and Tasmota will provide both values, plus others for both.  I have not see a big difference between the two other than the form factor and I like the S31 better for most things.

Jason2866, 2018-06-08:
@Frogmore42  I would like to see a European (Schuko) S31 Version. I think most people would like to use the POW as easy as a the S20...

ServerCare, 2018-06-11:
I actually wondered the same...  I have the POW as well as the POW R2 (which is more expensive). But frankly I do not see much difference ... I guess it is down to comparing the specs for the POW (HLW8012) and POW R2 (CSE7766 (alike)) chips...

ascillato, 2018-06-12:
Hi,

If your issue is solved, please close it. Thanks :+1:

closed 2018-06-12 by vaidyasr

EricVdB, 2018-06-29:
@Jason2866 there is a Mini Wi-Fi Socket in schuko version coming later this year (info from itead)

ascillato, 2018-07-23:
Sonoff POW vs POW R2, see comment https://github.com/arendst/Sonoff-Tasmota/issues/3232#issuecomment-407040337

xchg-dot-ca, 2021-12-06:
> I don't have a POW R2, but I have a POW and a S31 (which uses the same power measurement chip as the POW R2). They both measure current and voltage and Tasmota will provide both values, plus others [...]

Sorry for offtopic subject, but I looking for Original POW firmware, is there any change you can share yours?

Tinky1233, 2021-12-16:
> Sorry for off topic, but I looking for Original POW firmware, is there any change you can share yours?

Here is Sonof R2 Firmware: https://tinyurl.com/ut6pnz9d