arendst/Tasmota #2939: S31

tcpipchip, 2018-06-08:
Does the S31 has in pcb extras pins to add more sensors inside ? (i2c and Spi sensors and INFRARED for example) ?

znanev, 2018-06-08:
Even if it has, this device has measuring live voltage and current capabilities, which means that you don't have real ground and you can possibly fry anything you attach to it and/or electrocute yourself.

So my advice is - DO NOT connect any low-voltage stuff to S31's PCB if it is plugged into mains.

tcpipchip, 2018-06-08:
Ok, thank you!
Good should be use other  module to do this! right ?

znanev, 2018-06-08:
Exactly, you can use S20 (or its newest variant - S26). In S20 at least you can disable serial logging and reconfigure the Tx/Rx pins for I2C.

Just for the purpose of attaching additional sensors/electronics, stay clear of the devices with built-in energy measuring (like Sonoff POW or S31). Because their low-voltage electronics is not properly isolated from mains and can lead to nasty, if not deadly, surprises if abused!

digiblur, 2018-06-08:
The S20 version is great (US here), lots of room, easy to solder a header to and get to the pins.  Have to agree with the S31, super tight in there and would be difficult to put wires out to a sensor.  The S31 is great for what it does and its size but just don't add any sensors to it with the mains so close.

tcpipchip, 2018-06-09:
Nice! 
I see that S31 does not allow it to be upgrade by http....because reach 570K.
Is there some tip to make it ugradable ?

ascillato, 2018-06-10:
Hi @znanev , If your issue was solved, please close it. Thanks :+1:

znanev, 2018-06-11:
Hi @ascillato, I have only replied to @tcpipchip on this issue, so I cannot close it.

@tcpipchip - can you please close this issue if it has been resolved? Thanks!

closed 2018-06-11 by tcpipchip