arendst/Tasmota #2940: Make tcp connection to an ip

matrix200, 2018-06-08:
I want to peruse Sonoff Touch as an IoT button.
What I basically want to achieve is to define button press to ignore the relay and instead try to connect to a tcp port on my home server.
I understand this is not possible with the firmware right now (out of the box).
Am I right in my assumptions?

znanev, 2018-06-08:
Tasmota uses MQTT out of the box, so if you install a MQTT server (like Mosquitto) on your home server, then your Sonoff Touch will publish to it and you can subscribe whatever client you need to your MQTT server.

matrix200, 2018-06-08:
Yeah, but there are two points here:
1. My server is kind of busy (it runs quite a few apps and it is rather low power (J3455 cpu), so I would rather not install anymore things on it.
2.  It sounds like a kind of overkill just to install MQTT for such a minor thing.

znanev, 2018-06-08:
For IoT related tasks I would think that MQTT is kind of prerequisite. You can run one of the many MQTT server implementations even on low-powered hardware such as a Raspberry Pi. This will give you great flexibility in terms of publishing / subscribing to events you (or your home automation solution) are interested.

If you instead plan to directly communicate over TCP, then you have to first implement a TCP listener (server) yourself. Then you'll have to change the Tasmota firmware to be able to connect to your endpoint over TCP. But then you will have to implement your own protocol (i.e. the payload the TCP packets will transport). So it sounds like you want to reinvent the wheel. MQTT protocol, as well as MQTT client/server software is really lightweight, built with IoT in mind. Why not give it a try - I highly doubt you will notice any performance degradation on your home server.

matrix200, 2018-06-08:
Actually I already have it implemented (and there is no protocol).
Currently I use Amazon Dash buttons and a NAT redirect on my fw that redirects its requests to an xinetd listener that does nothing but simply run a script to perform the task (toggle various lights according to button pressed).

I will take your point under consideration though and try MQTT server out to see it affects my server.
Just to give you an idea what it runs:
1. Pfsense VM under KVM (for internet)
2. fileserver (samba)
3. pimatic (smart home)
4. ESET file server (for av scanning)
5. Apache (some misc stuff)
6. Mysql (related to work stuff).
7. Transmission server.

The server again is a J3455 four core with 8 gigs of RAM.

ascillato, 2018-06-08:
Please, read the wiki.

Tasmota already has REST api, so you can directly send commands to it.

MQTT is very light. For example I have Home Assistant running on a Raspberry Pi and, all the HA system + MQTT Server is using between 0% and 1% of the CPU.

Tasmota is mainly a MQTT device.

matrix200, 2018-06-08:
@ascillato, yeah I am aware of that, but I am not trying to send commands to it.
In fact I want the opposite (I want it to send commands to my server).
But like I said, I will give MQTT a try.

ascillato, 2018-06-08:
You have a `websend `command. For using that you should set rules by Tasmota console and config there your expected behavior.

matrix200, 2018-06-08:
Wow, thanks!
That was what I was looking for.
I searched through command reference,but apparently didn't do a very good job.

closed 2018-06-08 by matrix200