arendst/Tasmota #2941: Problems with esp8266 and relay module

umibozu83, 2018-06-08:
Hi I'm having problems with using esp8266( nodemcu v3) and relay module. The module is the traditional one of Arduino with 5v. Connect everything and it works great. The only flaw that if I turn on the nodemcu With the module powered The nodemcu does not start the firmware. Instead if you disconnect the power supply of the module (or stay powered off, disconnect the control pins) it starts up normally and reconnects the module and everything works.Some advice? Is this problem known? How can I solve?

umibozu83, 2018-06-08:
Sorry for bad english. Google XD translator

umibozu83, 2018-06-08:
Ps: The module has the possibility of being powered with external 5v For relay coils. Same problem.

Frogmore42, 2018-06-08:
How do you have things wired?
Most likely you are using one of the GPIOs that have to be a certain state to boot properly.

umibozu83, 2018-06-08:
I had also thought that now the problem is that I use all the outputs .... If I wanted to use only 4 relays and 4 buttons which should be used? It is better for me to limit myself to the relays because I have read That more than 4 are not configured in mqtt

umibozu83, 2018-06-08:
Otherwise I have to test each pin which one should use for the relays

Frogmore42, 2018-06-09:
Read this https://zoetrope.io/tech-blog/esp8266-bootloader-modes-and-gpio-state-startup/
Look up the conversion between esp8266 pin numbers and NodeMCU numbers.
Avoid the serial pins unless you know what you are doing.
Avoid GPIO 0 as a switch input.

Running 4 switches and 4 relays on the same NodeMCU will be tricky, as there are only so many pins and restrictions on quite a few of them.

If you are using a 5V relay board, it will also depend on if it has optocouplers or not and if not what voltage the inputs are at when not driven.

umibozu83, 2018-06-09:
ok done test ... wemos d1 r2 original does not make me this fault. function too well and much faster. Perhaps just the nodemcu from problems. In the future I will try the wemos d1 mini if ​​it does the same defect

ascillato, 2018-06-12:
Hi,

If your issue is solved, please close it. Thanks :+1:

closed 2018-06-27 by ascillato2