arendst/Tasmota #2942: IM GETTING A PROBLEM SETTING AN A NOMENCU 1.0 AS A CONTROLLER FOR SOME LED STRIP, FIRST I SET THE AMOUNG OF LEDS BUT IT CAN BE POSIBLE FROM THE COSOLE

wdavalil, 2018-06-08:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-06-08:
Please provide the information of the troubleshoting template. Without that we dont have enough information to help you. Also provide screenshots of the error.

reloxx13, 2018-06-08:
ANd pls stop shouting on us :D

wdavalil, 2018-06-09:
im very new at this just a couple weeks i start get intrested in this so i dont know what you are asking me i hope this can help.
[image]
[image]
[image]

wdavalil, 2018-06-09:
previusly i was trying to control different led strip with different gpios at the same time but there's only 07 ws2811 can this be done? i would like to control a cc stripat the same time with the same nodemcu

reloxx13, 2018-06-10:
this does not help-

read the wiki (commands)
provide result from command status 0
descripe your problem
edit the title

ascillato, 2018-06-10:
Hi @wdavalil 

You have deleted your previous messages. Can you provide error message you receive and all the information required in the template?

wdavalil, 2018-06-11:
I get an error. Now the NodeMCU doesn’t connect to the network when I
connect to consola you Can se some random caracteres and just keeps on. I
try get
Into reset but just keep doing the Same, before this I was trying to make
control 3 different stips but I can’t setup the same funsion in diferent
pins I’m
Not sure if there’s a way in consola or there’s is a better aproceh. Thanks
for
The quick answers I’m gonna attach of the status of the board in consola

ascillato, 2018-06-11:
Hi,

Please, need to be more specific. Which error? which type led strips?

----/Spanish Translation/----

Wilmer, nos tenés que dar mas datos. Con eso que pusiste no se entiende cual es el problema.
Que tipo de tira led es la que querés usar? si es una WS2812 solo se puede usar una en Tasmota, al menos por ahora. Si es de las tiras Led analógicas, podés usar varias con la opción PWM1, PWM2, etc

Otra cosa, la consola tiene una velocidad de transmisión (baudios) si está en otro valor distinto de 115200 vas a ver solo basura.

Tasmota tiene una pagina WEB cuando lo configurás y de ahí podes ver una consola y configurar los pines. Está todo explicado en la wiki (https://github.com/arendst/Sonoff-Tasmota/wiki)

Tasmota es un software para los dispositivos Sonoff para usarlo con MQTT para casas inteligentes. Si lo que vos querés es solo usar las tiras led, hay otros proyectos mas sencillos.

----/End Spanish Translation/----

wdavalil, 2018-06-11:
Estoy actualmente haciendo una lámpara de techo y estoy colocando unos led
ws2812  12mm digital rgb 5v. Y estoy configura do mi.casa con HASSio por eso
Me interesa este proyecto

ascillato, 2018-06-11:
Ok, y cual es el problema que tenés actualmente?

wdavalil, 2018-06-11:
La velocidad está settiada a 115200, son led ws2812 digital  12mm el
problema comenzó después de poner Tasmota en algunos SONOFF y ahora NodeMCU
cambie las cofifguracions siguiendo el wiki consola 18 se conectaba y había
comunicación MQTT pero solo controlaba 30 luces después de empezar a tratar
de configurar los números de led w tenía el circuito y a flashar varías
veces quede fuer de la red local en esos dispositivos

ascillato, 2018-06-11:
ah, ok, entonces parece que se te corrompió la configuración. Probaste tener el botón del NodeMCU apretado por 40 segundos? eso reseta toda la configuración a por defecto.

ascillato, 2018-06-11:
Ok, acordate que el chip ESP8266 no anda con 5G. solo 2.4Ghz. Si las 2 redes tienen el mismo nombre puede que ahí tengas el drama.

Si no te anda, fijate en la wiki que está explicado lo de borrar toda la flash con el programa esptool.py

Después para configurar el WS2812 es todo por comandos. Si lo haces via web es mas fácil. Desde el menú principal tenés el menú consola.

La cantidad de leds la definís con el comando pixel. Ah, ojo que la tira led usa 5V y bastante corriente así que necesita una fuente de 5V aparte. El NodeMCU no tiene suficiente corriente. Después la configuración para HASSIO está en la wiki.

Bueno, cuando puedas hacer las pruebas comentanos a ver como te fue.
Si podés editá el titulo a algo mas resumido.

Saludos.

wdavalil, 2018-06-11:
No creo porq está configurada para asignar a cada dispositivo el ancho de
banda que  necesita configure un módem aparte para la
Tarea  coloque un setup que tenía antes y los Sonoff y  NodeMCU se podía
conectar a la red antes igual cual branch debería descargar para tratar de
colocar el
Programa nuevamente yo tengo ahora development

ascillato, 2018-06-11:
> No creo porq está configurada para asignar a cada dispositivo el ancho de
> banda que  necesita

Ok, pero 5G y 2.4G son las frecuencias de las redes. Los Esp8266 NO FUNCIONAN en 5G, no tienen antena para eso.

> coloque un setup que tenía antes y los Sonoff y  NodeMCU se podía
> conectar a la red antes igual cual branch debería descargar para tratar de
> colocar el
> Programa nuevamente yo tengo ahora development

Wilmer, es dificil entender tu problema tanto en inglés como en español. Podrías redactarlo mejor y más específico?

Lo mejor siempre es usar la última versión, o sea la development.

closed 2018-06-12 by ascillato2