arendst/Tasmota #2944: Sonoff POW, overcurrent?

tomasroud, 2018-06-08:
Hi,

POW is working on USB correctly (Wifi, MQTT. all OK), but is dead when connected to a line.
I use POW to connect a boiler with 8A current max for ca one year.
A fuse on POW is interrupted, so there had to be more as 15A current, but the boiler has max 8A.
A circuit breaker in wall has max 16A and is still on.

I think short circuit had to be in POW only.
I am scary to change POW fuse :(

Have you ever seen this?
What can I do to avoid this situation?
[image]

Frogmore42, 2018-06-09:
That certainly looks no longer safe.  I would get a new one.  I would also double check that the load doesn't sometimes exceed 8A.  Also, is the load on for long periods, like hours?  That might be part of the problem.

Have you been keeping records of the load and can you see the peaks?

Jason2866, 2018-06-09:
One part of the board looks to me as had it contact with water (corrosion).....
Could this be?

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks.

closed 2018-07-10 by ascillato2

tomasroud, 2018-07-31:
The second one lasts 2 months. As you can see, the problem must be in the hardware. The same place is brown :( So, daily turn on and off lasts 2 months :( The harware of POW is not suitable to measure and switching a boiler with 8A current.
[image]
[image]

Jason2866, 2018-07-31:
If i where you i would switch with the POW a power relay which has contacts that can drive easily 16A or more...

tomasroud, 2018-07-31:
Ohhh man, that is the point of it!!!! Thank you very much. The consuption is stable, so I can calculate the time. Bye bye POW. Tomorrow I will do that with basic only. You are smarter as fox is. Thank you again. Problem solved. BTW still the POW is not recomended to switch 8A current daily.

Frogmore42, 2018-08-27:
I thought the picture looked familiar.  There was a recall of the POWs 

> here is a production defect in a small minority of Sonoff TH16 and POW that were produced from Dec 2016 to Jan 2017

comment from March 2017
https://community.openhab.org/t/itead-sonoff-switches-and-sockets-cheap-esp8266-wifi-mqtt-hardware/15024/134

Your picture looks exactly like the one that failed in that thread and for which they did the recall.  Yours sound like they are old enough to be from the batch.