arendst/Tasmota #2945: Having trouble giving SSID and erasing firmware

AalianKhan, 2018-06-08:
I am trying to flash a sonoff dual R2 with tasmota and I successfully flashed it but I am trying to give my SSID and password of my wifi through Termite (a serial communication program) but when I do shows this
`00:42:27 MQT: Attempting connection...
00:42:28 DNS: Query done. MQTT services found 0
00:42:28 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
ssid1 5920 Talisman
00:42:32 CMD: ssid1 5920 Talisman
00:42:32 RSL: stat/sonoff/RESULT = {"SSId1":"5920 Talisman"}
00:42:34 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset`
Then I tried to erase the firmware first by esp tool so I downloaded python and runned 

> Pip install esptool
And then I put my sonoff to programming mode and connect it and started to erase the firmware but when I typed the erase firmware command it said failed to process.
Please help

Jason2866, 2018-06-09:
Are you sure that you are in flash mode?

Jason2866, 2018-06-09:
Plz post the status messages of your erasing / flashing attempts

AalianKhan, 2018-06-09:
yes I put a female jumper wire at `BUTTON 0` and `GND` on my sonoff dual R2 and I also added the scripts you wanted

AalianKhan, 2018-06-09:
I redownloaded esp easy and

closed 2018-06-09 by AalianKhan