arendst/Tasmota #2946: Switchmode 1 - relay turns off and on every few seconds when button is depressed

law1964, 2018-06-08:
For my Sonoff 4ch Pro, I have set switch 3 and switch 4 in switchmode 1 so that the relay is on only when the button is depressed. When I press and hold the button, the relay turns on. After a few seconds, the relay turns off and on. This is repeated after another few seconds while the button still depressed.

- [x] Searched the problem in issues and in the wiki
- [x] Hardware used :  Sonoff 4ch Pro
- [x] Provide the output of command ``status 0`` :

21:22:57 RSL: STATUS = {"Status":{"Module":23,"FriendlyName":["Garage Door One","Garage Door Two","GD1isCLOSED","GD2isCLOSED"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:22:57 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Power on","Uptime":"0T00:21:49","StartupUTC":"2018-06-08T20:01:08","Sleep":0,"BootCount":10,"SaveCount":93,"SaveAddress":"F7000"}}
21:22:57 RSL: STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:29:54","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:22:57 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["neuronet",""],"TelePeriod":300,"SetOption":["00008001","55818000"]}}
21:22:57 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
21:22:57 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-2957","IPAddress":"192.168.20.35","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.1","Mac":"DC:4F:22:80:2B:8D","Webserver":2,"WifiConfig":2}}
21:22:57 RSL: STATUS7 = {"StatusTIM":{"UTC":"Fri Jun 08 20:22:57 2018","Local":"Fri Jun 08 21:22:57 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:47","Sunset":"20:51"}}
21:22:57 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-06-08T21:22:57"}}
21:22:57 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-06-08T21:22:57","Uptime":"0T00:21:49","Vcc":3.281,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"neuronet","RSSI":84,"APMac":"14:CC:20:B8:F2:85"}}}

**To Reproduce**
1. Enter "Switchmode3 1" and "Switchmode4 1" in the Console command line.
2. Press and hold Button 3 on the device.

**Expected behavior**
The relay should turn on and stay on as long as the button is depressed.

**Screenshots**
N/A

**Additional context**
I am controlling garage doors with Switch 1 and Switch 2. I want to connect reed switches to the GPIOs of Switch 3 and Switch 4 to use them to monitor open/closed status of the garage doors.

ascillato, 2018-06-08:
Hi,

Your config seems ok.

If you type in the console the command `pulsetime3`, it respond with zero?

law1964, 2018-06-08:
Thanks. I already thought of that because I had to set pulsetime for Switch1 and Switch2 to control the garage doors with a pulse of 0.3 seconds.

When I issue the commands pulsetime3 and pulsetime4, I get the following:
22:45:13 CMD: pulsetime3
22:45:13 RSL: RESULT = {"PulseTime3":"0 (Active 0)"}
22:45:19 CMD: pulsetime4
22:45:19 RSL: RESULT = {"PulseTime4":"0 (Active 0)"}

ascillato, 2018-06-08:
I tried to reproduce your issue on a NodeMCU configured as you have stated but it is working ok.

So, are your buttons working ok?

ascillato, 2018-06-08:
Can you type on the console `weblog 4` and try to reproduce that behavior again. Please post the console output.

law1964, 2018-06-08:
I think the buttons are fine. I tried a separate external toggle switch that I soldered parallel to the device's button and the results were the same.

Here is the output:

00:23:12 CMD: weblog 4
00:23:12 RSL: RESULT = {"WebLog":4}
00:23:12 CFG: Saved to flash at F8, Count 100, Bytes 2560
00:23:17 WIF: Checking connection...
00:23:17 WIF: Connected
00:23:31 APP: Button3 level 1-0
00:23:31 RSL: RESULT = {"POWER3":"ON"}
00:23:31 RSL: POWER3 = ON
00:23:31 CFG: Saved to flash at F7, Count 101, Bytes 2560
00:23:38 WIF: Checking connection...
00:23:38 WIF: Connected
00:23:44 APP: Button3 level 0-1
00:23:44 RSL: RESULT = {"POWER3":"OFF"}
00:23:44 RSL: POWER3 = OFF
00:23:45 CFG: Saved to flash at F6, Count 102, Bytes 2560
00:23:45 APP: Button3 level 1-0
00:23:45 RSL: RESULT = {"POWER3":"ON"}
00:23:45 RSL: POWER3 = ON
00:23:46 CFG: Saved to flash at F5, Count 103, Bytes 2560
00:23:52 APP: Button3 level 0-1
00:23:52 RSL: RESULT = {"POWER3":"OFF"}
00:23:52 RSL: POWER3 = OFF
00:23:52 CFG: Saved to flash at F4, Count 104, Bytes 2560
00:23:52 APP: Button3 level 1-0
00:23:52 RSL: RESULT = {"POWER3":"ON"}
00:23:52 RSL: POWER3 = ON
00:23:53 CFG: Saved to flash at FB, Count 105, Bytes 2560
00:23:54 APP: Button3 level 0-1
00:23:54 RSL: RESULT = {"POWER3":"OFF"}
00:23:54 RSL: POWER3 = OFF
00:23:54 CFG: Saved to flash at FA, Count 106, Bytes 2560
00:23:56 APP: Button3 level 1-0
00:23:56 RSL: RESULT = {"POWER3":"ON"}
00:23:56 RSL: POWER3 = ON
00:23:56 APP: Button3 level 0-1
00:23:56 CFG: Saved to flash at F9, Count 107, Bytes 2560
00:23:57 APP: Button4 level 1-0
00:23:57 RSL: RESULT = {"POWER4":"ON"}
00:23:57 RSL: POWER4 = ON
00:23:57 CFG: Saved to flash at F8, Count 108, Bytes 2560
00:23:58 WIF: Checking connection...
00:23:58 WIF: Connected
00:24:02 APP: Button4 level 0-1
00:24:02 RSL: RESULT = {"POWER4":"OFF"}
00:24:02 RSL: POWER4 = OFF
00:24:02 CFG: Saved to flash at F7, Count 109, Bytes 2560
00:24:03 APP: Button4 level 1-0
00:24:03 RSL: RESULT = {"POWER4":"ON"}
00:24:03 RSL: POWER4 = ON
00:24:03 CFG: Saved to flash at F6, Count 110, Bytes 2560
00:24:10 APP: Button4 level 0-1
00:24:10 RSL: RESULT = {"POWER4":"OFF"}
00:24:10 RSL: POWER4 = OFF
00:24:10 CFG: Saved to flash at F5, Count 111, Bytes 2560
00:24:11 APP: Button4 level 1-0
00:24:11 RSL: RESULT = {"POWER4":"ON"}
00:24:11 RSL: POWER4 = ON
00:24:11 CFG: Saved to flash at F4, Count 112, Bytes 2560
00:24:12 APP: Button4 level 0-1
00:24:19 WIF: Checking connection...
00:24:19 WIF: Connected
00:24:23 APP: Button3 level 1-0
00:24:23 RSL: RESULT = {"POWER3":"OFF"}
00:24:23 RSL: POWER3 = OFF
00:24:23 APP: Button3 level 0-1
00:24:24 CFG: Saved to flash at FB, Count 113, Bytes 2560
00:24:24 APP: Button4 level 1-0
00:24:24 RSL: RESULT = {"POWER4":"OFF"}
00:24:24 RSL: POWER4 = OFF
00:24:25 APP: Button4 level 0-1
00:24:25 CFG: Saved to flash at FA, Count 114, Bytes 2560
00:24:30 APP: Button3 level 1-0
00:24:30 RSL: RESULT = {"POWER3":"ON"}
00:24:30 RSL: POWER3 = ON
00:24:31 CFG: Saved to flash at F9, Count 115, Bytes 2560
00:24:39 WIF: Checking connection...
00:24:39 WIF: Connected
00:24:42 APP: Button3 level 0-1
00:24:42 RSL: RESULT = {"POWER3":"OFF"}
00:24:42 RSL: POWER3 = OFF
00:24:43 APP: Button3 level 1-0
00:24:43 RSL: RESULT = {"POWER3":"ON"}
00:24:43 RSL: POWER3 = ON
00:24:44 APP: Button3 level 0-1
00:24:44 RSL: RESULT = {"POWER3":"OFF"}
00:24:44 RSL: POWER3 = OFF
00:24:45 CFG: Saved to flash at F8, Count 116, Bytes 2560
00:24:59 WIF: Checking connection...
00:24:59 WIF: Connected
00:25:20 WIF: Checking connection...
00:25:20 WIF: Connected
00:25:40 WIF: Checking connection...
00:25:40 WIF: Connected
00:26:00 WIF: Checking connection...
00:26:00 WIF: Connected

Another observation: I tried switchmode 1 with a sonoff basic and there was no effect. It behaved the same as switchmode 0 or 4. The relay would not toggle on until I released the button.

law1964, 2018-06-09:
I have opened another issue #2954 for the Switchmode on the Sonoff Basic. The symptoms are different from this problem on the 4CH Pro.

law1964, 2018-06-10:
@ascillato Thank you for the help. I wasn't able to resolve the problem with using the device's pushbutton with Switchmode 1. However, it is working with an external switch connected to the RX pin and linked to Switch 3. I did the same with another external switch and the TX pin linked to Switch 4.

I am pleased that I can implement the reed switch sensor for the garage door OPEN/CLOSED state, but I am still puzzled as to why I was getting the toggling with the native pushbutton and a switch connected to the contacts of the pushbutton.  The 4CH PRO has a secondary STM32 that handles the pushbuttons and the relays? Is it possible that the extraneous toggling was caused by this STM32 microcontroller?

ascillato, 2018-06-10:
Hi

Sorry. I don't know. I don't have a 4ch Pro to test that. May be someone else can help us with this.

closed 2018-08-09 by stale[bot]