arendst/Tasmota #2950: S20 1.3 EU version no WIFI AP after flashing

casperse, 2018-06-09:
**Describe the bug**
On the esp chit itself is just printed that it is a 8266.
After flashing with the newest Tasmota there is no WIFI AP after flashing.
Many post on the internet confirms this:
https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3615&sid=b028928158d94160d8640088a5e66315&start=10

I found that using the latest ESPEasy version and flashing it using the FlashESP8266.exe and selecting the 8285 firmware (ESPEasy_v2.0.0-dev12_normal_8285.bin). would resolve the problem

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues and in the wiki
- [X] Hardware used : Sonoff S20 EU version 1.3 On the esp chit itself is just printed that it is a 8266. 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-06-09:
Do you have flashed with Option DOUT? That is the reason why ESPeasy 8285 works....
**The problem you describe is not TASMOTA related. It occours only when flashing is done wrong.**
Flash sonoff always as described in the wiki.
S20 works without any problems.

casperse, 2018-06-09:
Sorry I used the flashing tool "borrowed" from ESPeasy package like recommended on all the youtube videoes and there is no setting just selecting the bin file....so you are saying that this tool dosent flash TASMOTA correctly? Sorry new to this...have flashed basic and POW this way without issues

Jason2866, 2018-06-09:
There are many **wrong** infos around...
Just read the wiki from arendst here and you will get all your devices working.
You where lucky that your other attempts work

Jason2866, 2018-06-09:
Important User Compilation Information
If you want to compile Sonoff-Tasmota yourself keep in mind the following:

Only Flash Mode DOUT is supported. Do not use Flash Mode DIO / QIO / QOUT as it might seem to brick your device. See Wiki for background information.
Sonoff-Tasmota uses a 1M linker script WITHOUT spiffs 1M (no SPIFFS) for optimal code space. If you compile using ESP/Arduino library 2.3.0 then download the provided new linker script to your Arduino IDE or Platformio base folder. Later version of ESP/Arduino library already contain the correct linker script. See Wiki > Prerequisite.
To make compile time changes to Sonoff-Tasmota it can use the user_config_override.h file. It assures keeping your settings when you download and compile a new version. To use user_config.override.h you will have to make a copy of the provided user_config.override_sample.h file and add your setting overrides. To enable the override file you will need to use a compile define as documented in the user_config_override_sample.h file.

casperse, 2018-06-09:
Do you know the installer? FlashESP8266.exe its actual making it possible for everyone to flash Sonoff's without much effort... (But I will look into the wicki)

casperse, 2018-06-09:
Sorry most of what you are telling me is beyound my skill levels

Jason2866, 2018-06-09:
No i dont know this installer. 
It is not a good idea to use on sonoff devices because you have got a non working device...

You dont have to understand it all. Just do it as in wiki described every other way is not tested confirmed to work. It could work... But you will get no support or you have to answer questions you dont know.

So the easy successfull way is doing it in the well approved way as described in the wiki

casperse, 2018-06-09:
OK GOT IT - BUT in your WIKI you actual link directly to DrZzs Youtube Channel - Great and Easy Tasmota related Videos and in this video and below he tells everyone about this new way of flashing Sonoffs...just saying :-)

Jason2866, 2018-06-09:
Yep, you could do an issue that the way the video describes doesnt results always in a successfull flashed device.

casperse, 2018-06-09:
Ok did it like the wiki said and:
Generic ESP 8266 module
Flash mode DOUT
1 M No SPIFFS
1.4 prebuilt
26mhz
40mhz
80mhz
115200 (upload speed)

And everything compiled ok flashed to the Sonod EU S20 and still no wifi access point same as before
Using arduino-1.8.5 and ESP8266 Board version 2.4.0...

casperse, 2018-06-09:
I used exactly the same setup on a TH16 and that worked perfectly!
There is something different with the Sonoff S20 EU model sofar the only thing working flashing that is ESPeasy 8285 only explanation that Ardiono wiki flash works for one and not the other

casperse, 2018-06-09:
Also working on Sonoff Pow R2! - Something special about the S20 EU model, only able to flash it with ESPeasy... any input to what I could try to find out why Tasmota isnt working on S20 (I have two of them both give same result)

Jason2866, 2018-06-09:
Use Arduino 2.3.0.   Version 2.4.0 has bugs
See https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips

casperse, 2018-06-09:
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE shows setup for 2.4.0 but again I just flashed all of these Sonoff devices using 2.4.0 with the recommended settings worked perfectly.
Also I can not select 1M No SPIFFS on 2.3.0?

Jason2866, 2018-06-09:
Wiki...
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

Jason2866, 2018-06-09:
No your S20 do not work with 2.4.0.

casperse, 2018-06-09:
I have done that and copied the Replace in Arduino IDE folder portable\packages\esp8266\hardware\esp8266\2.3.0 file boards.txt with the Tasmota Source code file arduino\version 2.3.0\boards.txt.

That does not provide me with the option 1M No SPIFFS only 512!

casperse, 2018-06-09:
ok If not available copy from the Tasmota release Source code folder arduino\version 2.3.0\tools\sdk\ld file eagle.flash.1m0.ld to Arduino IDE folder portable\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\ld.
that did it after a reboot... still not having wifi AP after flashing?

Frogmore42, 2018-06-09:
serial log and status 0?

reloxx13, 2018-06-10:
their is no wifi ap on default.

default is wps mode.

did you change this?

casperse, 2018-06-11:
Sorry after 12 hours of updating firmware all with success except the DAAAAM S20!
I managed to destroy the USB device used for flashing. (So I am awaiting a replacement)
Sorry I have tried both standard flash and 4 push to the reset button in order to access it.
Also tried changing the config before flashing both worked on the POW R2 model and the TH16 and basic models... 

I have two of these S20 and I really don't want anything other than Tasmota on them :-)
Again thanks for your support - I will provide any information you want when I have a new USB flash device

casperse, 2018-06-14:
Just a thought the buttom function is listed here: https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage but since the S20 is also functioning as a power On/Off button does that change anything? like 4 short presses is actual 5?

closed 2018-07-10 by ascillato2