arendst/Tasmota #2953: When trying to upload flash code to SOnOff basic

fritserasmus, 2018-06-09:
- [ ] Hardware used : Window 7 64
- [ ] Provide the output of command ``status 0`` :
```
[06/09/18 17:18:19] Processing sonoff-minimal (platform: espressif8266; board: esp01_1m; framework: arduino)
 PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
SYSTEM: ESP8266 80MHz 80KB RAM (1MB Flash)
Converting sonoff.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP KNX IP Library> v0.5
|   |-- <EEPROM> v1.0
|   |-- <ESP8266WebServer> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266HTTPClient> v1.1
|   |-- <ESP8266WiFi> v1.0
|-- <SPI> v1.0
|-- <Adafruit SGP30 Sensor> v1.0.2
|   |-- <Wire> v1.0
|-- <OneWire> v2.3.2
|-- <NeoPixelBus> v2.2.9
|   |-- <SPI> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0
|-- <Mutichannel_Gas_Sensor> v0.0.1
|   |-- <Wire> v1.0
|-- <ESP8266httpUpdate> v1.1
|   |-- <ESP8266HTTPClient> v1.1
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WiFi> v1.0
|-- <TasmotaSerial> v1.3.0
|-- <TasmotaMqtt> v1.0.0
|-- <PubSubClient> v2.6
|-- <Wire> v1.0
|-- <ESP8266WebServer> v1.0
|   |-- <ESP8266WiFi> v1.0
|-- <ArduinoOTA> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|-- <DNSServer> v1.1.0
|   |-- <ESP8266WiFi> v1.0
|-- <IRremoteESP8266> v2.2.1
|-- <ArduinoJson> v5.11.2
|-- <Ticker> v1.0
|-- <I2Cdevlib-ADS1115>
|   |-- <I2Cdevlib-Core>
|   |   |-- <Wire> v1.0
|-- <Adafruit BME680 Library> v1.0.5
|   |-- <Adafruit Unified Sensor> v1.0.2
|   |-- <Wire> v1.0
|   |-- <SPI> v1.0
|-- <ESP MQTT> v1.0.1
|-- <Joba_Tsl2561> v2.0.1
|   |-- <Wire> v1.0
|-- <ESP8266WiFi> v1.0
Compiling .pioenvs\sonoff-minimal\src\core_esp8266_timer.c.o
Compiling .pioenvs\sonoff-minimal\src\core_esp8266_wiring_digital.c.o
Compiling .pioenvs\sonoff-minimal\src\core_esp8266_wiring_pwm.c.o
...................====>>
I have excluded some lines here as there was no issues in this.
...................====>>
Compiling .pioenvs\sonoff\FrameworkArduino\umm_malloc\umm_malloc.c.o
Archiving .pioenvs\sonoff\libFrameworkArduino.a
Linking .pioenvs\sonoff\firmware.elf
Checking size .pioenvs\sonoff\firmware.elf
Building .pioenvs\sonoff\firmware.bin
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [======    ]  64.1% (used 52536 bytes from 81920 bytes)
PROGRAM: [=====     ]  51.9% (used 543893 bytes from 1048576 bytes)
Configuring upload protocol...
Looking for upload port...
Use manually specified: COM5
Uploading .pioenvs\sonoff\firmware.bin
**error: Failed to open COM5**
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 2
 [ERROR] Took 54.90 seconds

```

**(Please, remember to close the issue when the problem has been addressed)**

The error message is: error: Failed to open COM5

I suppose it comes from line 66 of platformio.ini

I tried in Windows Device manager to find the COM port, but was not showing there

Any help or pointers would be appreciated

I even tried to run the "esptool usage"
on 
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting

Jason2866, 2018-06-09:
You have to change in platformio.ini the setting to the correct COM Port.
This means to the COM Port your Adapter uses. If you dont see here one in the Device Manager
than your drivers for the adapter are not correct installed.

fritserasmus, 2018-06-09:
:-( 
The USB Cable I used was causing the problem!!!!!
Problem solved when using a good cable

closed 2018-06-09 by fritserasmus

jeffl78, 2018-12-19:
I'm having this issue with widows 8.2 on an Asus laptop.  I bought a fresh cable to rule that out the other day but I get the same error.