arendst/Tasmota #2958: RF Bridge and Node-Red. Help Please

Slowsteve, 2018-06-10:
RF Bridge version 5.14.0
I have learnt code for key16. The console shows an entry each time the remote button is pressed:-
12:52:57 MQT: tele/sonoff_bridge/RESULT = {"RfReceived":{"Sync":5920,"Low":190,"High":590,"Data":"DE2EE3","RfKey":16}}
12:53:01 MQT: tele/sonoff_bridge/RESULT = {"RfReceived":{"Sync":5910,"Low":190,"High":590,"Data":"DE2EE3","RfKey":16}}
12:53:04 MQT: tele/sonoff_bridge/RESULT = {"RfReceived":{"Sync":5910,"Low":190,"High":590,"Data":"DE2EE3","RfKey":16}}
12:53:07 MQT: tele/sonoff_bridge/RESULT = {"RfReceived":{"Sync":5920,"Low":190,"High":590,"Data":"DE2EE3","RfKey":16}}
12:53:15 MQT: tele/sonoff_bridge/RESULT = {"RfReceived":{"Sync":5920,"Low":190,"High":590,"Data":"DE2EE3","RfKey":16}}
12:53:22 MQT: tele/sonoff_bridge/RESULT = {"RfReceived":{"Sync":5910,"Low":190,"High":590,"Data":"DE2EE3","RfKey":16}}
12:53:33 MQT: tele/sonoff_bridge/RESULT = {"RfReceived":{"Sync":5910,"Low":190,"High":590,"Data":"DE2EE3","RfKey":16}}
The "Sync" value changes but the other fields seem constant. 
I can subscribe in mosquitto and see the same entries.
**I would like to make a flow in node-red that starts with a mqtt input listening for "RfKey":16**
(My intended usage is to send notification and flash house lights when the rf doorbell is pressed)
I'm very new to all this and would appreciate guidance or an example of how to configure the node settings.(Topic,Message,Payload?) I have spent many hours trying to find an answer but cannot get the correct "syntax"?

MatzeJoerling, 2018-06-10:
`[{"id":"18418d89.24d9c2","type":"mqtt in","z":"a6230776.dc0908","name":"","topic":"tele/sonoff_bridge/RESULT","qos":"2","broker":"5fcd8a44.3ec094","x":360,"y":2060,"wires":[["d1681ac0.0b5d58"]]},{"id":"dee3c45f.a9ee38","type":"inject","z":"a6230776.dc0908","name":"RFKey16","topic":"tele/sonoff_bridge/RESULT","payload":"{\"RfReceived\":{\"Sync\":5930,\"Low\":190,\"High\":590,\"Data\":\"DE2EE3\",\"RfKey\":16}}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":420,"y":2000,"wires":[["61831d07.4371b4","d1681ac0.0b5d58"]]},{"id":"d1681ac0.0b5d58","type":"switch","z":"a6230776.dc0908","name":"","property":"payload.RfReceived.RfKey","propertyType":"msg","rules":[{"t":"eq","v":"16","vt":"num"},{"t":"else"}],"checkall":"true","repair":false,"outputs":2,"x":650,"y":2060,"wires":[["d81e7047.5871","828874b1.1f5968"],[]]},{"id":"61831d07.4371b4","type":"debug","z":"a6230776.dc0908","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":650,"y":2000,"wires":[]},{"id":"d81e7047.5871","type":"function","z":"a6230776.dc0908","name":"","func":"// DO WHATEVER YOU WANT\n// ONLY RFKEY16 will arrive here\nnode.status({fill:\"green\",shape:\"ring\",text:\"RFKey 16 pressed\" });\n\nreturn msg;","outputs":1,"noerr":0,"x":830,"y":2040,"wires":[[]]},{"id":"828874b1.1f5968","type":"debug","z":"a6230776.dc0908","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":830,"y":2000,"wires":[]},{"id":"f0b8b42e.087818","type":"inject","z":"a6230776.dc0908","name":"RFKey15","topic":"tele/sonoff_bridge/RESULT","payload":"{\"RfReceived\":{\"Sync\":5930,\"Low\":190,\"High\":590,\"Data\":\"DE2EE3\",\"RfKey\":15}}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":420,"y":1960,"wires":[["61831d07.4371b4","d1681ac0.0b5d58"]]},{"id":"5fcd8a44.3ec094","type":"mqtt-broker","z":"","name":"localhost","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""}]`

[image]

Check it out, the function node can do whatever you program when RF16 is pressed. 
Watch Debug Tab when running...

Slowsteve, 2018-06-11:
MatzeJoerling You're a star! I've set it up with just  mqtt-in, json, switch and  mqtt-out nodes.
So, if I understand this correctly, the first node listens for anything with the topic "tele/sonoff_bridge/RESULT",the json node translates then the switch fires if the message has RfKey value of 16? Works brilliantly! For my education, I modified the switch replacing "RfKey" with" Data" and "16" with "DE2EE3" ,this also works. Does this mean that multiple conditions could be added? If so, can they be arranged as and/or (both/either ; all/any)?
Thanks for your help, I'm so chuffed to be making progress at last.

MatzeJoerling, 2018-06-11:
Hi, nice to hear ;-)
Switch ony works on one attribute and the message takes that output! you could sure put 3 or 4 switch nodes after another and switch what you want. but if you want AND/OR logical gate i would use node-red-contrib-bool-gate . There you can work with multiple messages and logic functions (and / or / xor ...). 

to be more specific you have to give me more details ...

closed 2018-07-10 by ascillato2