arendst/Tasmota #2960: Sonoff 4CH Pro turning off other relays after turning 1 on

pascallieverse, 2018-06-10:
Hello,

When I want to turn on a channel on my Sonoff 4CH Pro it automatically turn’s off the other channels. I just don’t get it to work with more than one channel turned on. I have used the web interface and MQTT to send those commends.

The log I am getting is:
```
18:27:08 MQT: stat/pro/RESULT = {"POWER1":"ON"}
18:27:08 MQT: stat/pro/POWER1 = ON
18:27:11 MQT: stat/pro/RESULT = {"POWER2":"ON"}
18:27:11 MQT: stat/pro/POWER2 = ON
18:27:11 MQT: stat/pro/RESULT = {"POWER1":"OFF"}
18:27:11 MQT: stat/pro/POWER1 = OFF
```

Is it that the 4CH Pro can only handle 1 channel powered on with Tasmota?

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : Sonoff 4CH Pro Module
- [X] Provide the output of command ``status 0`` :
```
18:37:06 MQT: stat/pro/STATUS = {"Status":{"Module":23,"FriendlyName":"Sonoff","Topic":"pro","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:37:06 MQT: stat/pro/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:10:06","Sleep":0,"BootCount":11,"SaveCount":53,"SaveAddress":"F7000"}}
18:37:06 MQT: stat/pro/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:37:06 MQT: stat/pro/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"Lieverse","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
18:37:06 MQT: stat/pro/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
18:37:06 MQT: stat/pro/STATUS5 = {"StatusNET":{"Hostname":"pro-3257","IPAddress":"192.168.88.207","Gateway":"192.168.88.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.88.1","Mac":"60:01:94:85:6C:B9","Webserver":2,"WifiConfig":3}}
18:37:06 MQT: stat/pro/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.88.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_856CB9","MqttUser":"admin","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:37:06 MQT: stat/pro/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 10 17:37:06 2018","Local":"Sun Jun 10 18:37:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
18:37:06 MQT: stat/pro/STATUS10 = {"StatusSNS":{"Time":"2018-06-10T18:37:06"}}
18:37:06 MQT: stat/pro/STATUS11 = {"StatusSTS":{"Time":"2018-06-10T18:37:06","Uptime":"0T00:10:06","Vcc":3.210,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Lieverse","RSSI":78,"APMac":"E4:8D:8C:84:C3:00"}}}
18:37:14 MQT: tele/pro/STATE = {"Time":"2018-06-10T18:37:14","Uptime":"0T00:10:14","Vcc":3.210,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Lieverse","RSSI":88,"APMac":"E4:8D:8C:84:C3:00"}}
```

Snipercaine, 2018-06-10:
the Pro has an "interlocking" function, but you can change it to "self-locking":
https://youtu.be/NNW05smSNOA?t=10m21s

pascallieverse, 2018-06-10:
Really thanks!!! It works now, I guess I was just searching on the wrong words.

closed 2018-06-10 by pascallieverse