arendst/Tasmota #2961: Having trouble giving SSID and erasing firmware

AalianKhan, 2018-06-10:
I successfully flash my Sonoff dual R2 with Tasmota but I am trying to give my SSID and password of my wifi through Termite (a serial terminal program) but when I do, it shows the following

> 00:42:27 MQT: Attempting connection...
00:42:28 DNS: Query done. MQTT services found 0
00:42:28 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
ssid1 5920 Talisman
00:42:32 CMD: ssid1 5920 Talisman
00:42:32 RSL: stat/sonoff/RESULT = {"SSId1":"5920 Talisman"}
00:42:34 APP: Restarting
ets Jan  8 2013,rst cause:1, boot mode:(1,7)
ets Jan  8 2013,rst cause:4, boot mode:(1,7)
wdt reset

to me, it looks like the `wdt reset` messes up everything up. when `wdt reset` shows up, my Sonoff resets and never turns on again and when I reconnect by my FTDI adapter 1 blue flash blinks and no response comes through Termite. Then I tried reinstalling Tasomota which was successful and tried to give my SSID again but nope, still same problem (and script). here is a detailed video of what I am doing 
> https://drive.google.com/open?id=1ZogBtEgUu9WnKSyxFoIsu5yDlcRsWdNx

Then I tried to erase the firmware first by esp tool so I downloaded python and ran the command 

`pip install esptool`

And then I put my Sonoff to programming mode and connect it and started to erase the firmware but when I typed the erase firmware command it said failed to process.

> Microsoft Windows [Version 10.0.16299.461]
(c) 2017 Microsoft Corporation. All rights reserved.
C:\Users\Aalian khan\AppData\Local\Programs\Python\Python36-32\Scripts>esptool.py --port COM8 erase_flash
failed to create process.
C:\Users\Aalian khan\AppData\Local\Programs\Python\Python36-32\Scripts>_

Please help

AalianKhan, 2018-06-10:
I have researched a bit and read that the power supply isn't enough but I'm not sure about that because I have flashed other Sonoff dual and they are working fine.

ascillato, 2018-06-11:
00:00:00 WIF: Connecting to AP1 **5920 Talisman** in mode 11N as sonoff-8018...

Space in SSID = RESET

ascillato, 2018-06-13:
Hi,

Could you manage to solve it?

AalianKhan, 2018-06-13:
Thank you very much it connected to the internet but there is like a 5 second delay

AalianKhan, 2018-06-13:
thanks your the best. may your problems in life be solved because you solved mine 🥇

ascillato, 2018-06-13:
Great!

Happy to know that it is working for you. :+1: 

Regards

closed 2018-06-13 by ascillato2