arendst/Tasmota #2962: Switch off all timers with one command

witek1308, 2018-06-10:
Hi,
I couldn't find this feature in available commands but would be great to have single command to switch off all configured timers. Use case: controlling domestic hot water circulation pump or water sprinklers.
I want to configure number of programs to switch on/off circulation pump when required. I want to be able however to switch off all timers for holidays (out of home) not to disarm them one by one but make sure nothing will switch on (or off) my system during my absence.

ascillato, 2018-06-10:
Hi

Nice idea :+1: 

Something like `timers 0` to disarm all timers?

witek1308, 2018-06-10:
Exactly!

closed 2018-06-26 by arendst

witek1308, 2018-06-26:
After reading the commit (which I hope I interpreted properly) actually I mentioned something else: I intended to have main flag (Timers 0/1) only after which being set/unset, timers are being checked and executed. This will allow for configuring timers once but disabling it when needed. Then enable would be simple not one by one again. From the customer perspective this would implement situation when timers are arbitrally disabled (e.g. on holidays) and enabled again when back to normal.

arendst, 2018-06-26:
I knew it. It was either this or the other...