arendst/Tasmota #2963: High power consumption of Tasmota

qingz2004, 2018-06-10:
I have a few Sonoff S31. Two of them flashed with Tasmota 5.14.0a. They're all connected to my WiFi network.
I used an AC power meter to measure the power consumption of the S31s with no load (nothing plugged in).
The one with original Sonoff firmware showed 0W, no matter switch is on or off.
But with the two S31s flashed with Tasmota, I got 0.9-1.0W if switch is on and 0W if switch is off.

I don't understand why S31s with Tasmota take much more power than they with the stock firmware if the relay is closed.

reloxx13, 2018-06-10:
which library did you use?

and read teh wiki:
https://github.com/arendst/Sonoff-Tasmota/wiki/Energy-Saving

qingz2004, 2018-06-10:
I have Arduino 2.3.0, 2.4.0 and 2.4.1 installed, but I think I'm using the default 2.3.0 since platform = espressif8266@1.5.0.
S31 has power monitor itself, it needs check and report power continuously. Don't know if I can set sleep time.
Why it did not take any power or much less power when the relay is off? It has nothing to do with sleep mode.

Thanks!

Phrede, 2018-06-10:
Consider the accuracy of your measurements.  Obviously, it is not 0 watts ever since nothing functions without some power.  Then consider the scale of measurements and the impact of such a small amount.  Remember, there is a mechanical relay inside the Sonoff and that the coil of the relay consumes power when on.

qingz2004, 2018-06-10:
I knew 0W is not really 0W, but it's definitely less than 0.1W or it will display 0.1W instead of 0.0W.
So when relay is on, the stock firmware consume less than 0.1W while Tasmota consumes more than 0.9W.
When relay is off, both consume less than 0.1W.
I don't understand what is happening there.

ascillato, 2018-06-10:
Hi,

Are you 100% sure that with the stock firmware is 0W when you switch the relay ON?

1W in 220V is 4.5mA of current

The coil of the relay takes about 5mA of current.

So, the 0.9-1W when it is ON, is the expected value.

qingz2004, 2018-06-10:
Yes. That is another thing which I don't understand.
I knew the coil inside relay would drain around 50mA at 5V typically, which is around 0.25W.
I stared at the power meter for more than 1 minute, and it keeps 0.0W all the time with stock firmware, relay was on.
With Tastoma and with sleep time not 0, the meter shows 0.0W and 0.9W alternatively.
So the relay should drain less than 0.1W, or my meter has some offset.
But with the same meter, the stock and Tasmota have 0.9W the difference.

ascillato, 2018-06-11:
I was looking for the schematics of the S31 to see if it has some sort of knowing if something is attached to its plug (because if it does, may be, the stock firmware was not turning the relay on) but seems that S31 does not have that feature.

So, I don't know what is happening in your setup, but if the relay is ON, it will consume about 50mA of DC current and plus the esp8266 power, so is not going to be 0W.

If you want to lower the stand-by power, you need to use the sleep command. You said that you want the power to be reported continously. The sleep command will not disturbe that. For example, If you are using Tasmota flashed to be at 80Mhz, you will have about 30000 free main loop cycles per second. So, Tasmota can check things (besides wifi and the rest) a lot of times per second.
When using sleep like 100, you have about 10 free main loop cycles per second. So, Tasmota also can send each updates of your power, and the power consumption of the ESP8266 will be low.

Frogmore42, 2018-06-11:
What are you using for a meter and how accurate is it, especially at low power levels and non standard power factors.  The power supply in Sonoffs is a low cost one and likely has poor power factor and unless you have a very good power meter you will not have a good enough picture to make a good decision.

Normal latching relays are about twice the price and complexity, so probably not even available from the really low cost suppliers.  They would likely need additional code and hardware to work safely.  Most people wouldn't want to pay more for the modest savings in power at todays power rates.  Assuming only $1 extra cost, the savings of a latching relay would take about 10 years at 3-4hours/day usage to pay back.  I don't believe that would be worthwhile.

qingz2004, 2018-06-11:
I don't need the power meter to be accurate. I use it to be a trigger in the rule, and send me notifications. So I know the washer is done, or my kid is watching TV instead of doing homework...
At the same time, I also use it to report the room temperature and humidity. Of course, I use isolated sensors.
If the difference is 0.9W, it will take 0.65kWh/month. If I have 10 of these, then 6.5kWh/month is not small part in my hydra bill.
I can not short the relay output to make it always close since it's on the other side of the PCB.

Frogmore42, 2018-06-11:
My question about the power meter accuracy, was the one you used to measure the power utilization of the S31.  The S31 is fine to use as an indication of usage, but is not really that accurate and probably worse at low levels.

You can short the relay output.  If you search the issues list, you will see that a few people who have already done some.  There is also an option to still measure power when the switch/relay is off.

If your goal is to just measure power utilization and NOT to control the load, using the POW/S31 is NOT the best choice. I use the iotawatt https://community.iotawatt.com/
Currently doesn't support MQTT directly, but it is in the plan and it could be done with NodeRED talking to emoncms or influx.  One iotawatt has 14 channels and uses about 1-2W and is much more accurate than the POW/S31.

I am not sure what your normal hydra bill is or what electricity costs in your part of the world.  In my part of the world, the incremental cost is about $0.12/kWh.  So 6.5kWh is $0.78.  Since I have an electric water heater, daily usage is 20-40kWh so 6.5/month is a fairly insignificant part of my bill

qingz2004, 2018-06-11:
The accuracy of my power meter is 0.1kWh.

> You can short the relay output. If you search the issues list, you will see that a few people who have already done some. There is also an option to still measure power when the switch/relay is off.

I searched the issues list, can't find how to short the relay output, and where is the option to measure power when the switch/relay is off.
Would you please point it out?
Thanks!

ascillato, 2018-06-11:
https://github.com/arendst/Sonoff-Tasmota/issues/1420

qingz2004, 2018-06-11:
Thanks!
SetOption21 1 to always allow energy monitoring.
But I still can't short the relay on S31, since it's on the other side of PCB. It is hard to de-solder and get access to it.

ascillato, 2018-06-12:
Hi,

If your issue is solved, please close it. Thanks :+1:

qingz2004, 2018-06-12:
I'm measuring the how much energy it consumes in 24 hours when relay is or and off and compare it against the stock firmware.
Will report back in 4 days.

Phrede, 2018-06-14:
Measuring over time is a good idea to get a better idea of real world costs.  Don't forget the power factor.  If this has a major impact on your power costs then I expect your utility company will be picky about the current leading or lagging.

FWIW, your example would cost me $0.51 per month in energy costs at my highest rates not taking into account any time or tier discounting.

qingz2004, 2018-06-14:
I try to understand why there is so much difference on power consumption between tasmota and stock firmware when relay is on. And try to make sure this is true, and not the measurement makes the difference. 
The energy cost is not a big deal. But it will be a big deal for battery powered devices.

Frogmore42, 2018-06-14:
@qingz2004 you are confusing resolution and accuracy.  Your meter has a resolution of 0.1kW?  Even assuming it is 0.1W resolution, that does not indicate the accuracy.  Unless you spent thousands of dollars on it, it will (at best) have an accuracy of 2-3% +/1 or more counts.  A typical good DMM has an accuracy spec of +/- 2% for ACV RMS.  That means if your actual voltage is 240V, it might read 235-245 and be within spec.  Current is harder to measure accurately (since it is typically done by converting it to a voltage by a resistor, which has it's own accuracy spec).  To measure real loads, it has to measure 10s of amps, but your load (at 240V, 1W is about 4mA).  So, let's say it can measure 20A. 4mA is 1/5000th of that.  Getting that type of range with a 16bit ADC is all but impossible.  When you thrown in the non-linear loads that these types of power supplies present, you are not going to get numbers that are anywhere near believable.  

They might be interesting and they might even be repeatable, but I don't believe they are really an accurate picture.  (your meter probably has a level below which it will read 0, real revenue grade meters for houses typically don't start registering power until the usage gets above 5W). 

I don't doubt that there is a difference in power utilization of the Tasmota FW and the stock one.  Remember that Tasmota is set up for MQTT and it will depend on what teleperiod you have set and what sleep if any you have set.  The stock FW has a different set of requirements and might have made some effort to put the esp8266 to sleep most of the time.

qingz2004, 2018-06-14:
The spec says measurement accuracy is 1%. I don't know the resolution, but it can display 0.1W and 0.0001kWh.
I'm using the same meter to measure the differences, so the accuracy does not matter that much, only resolution matters.
You are right, since the current is so small, if the resolution is not high enough, the result, even accumulated, may not mean anything.
Looks like I need an AC current meter with resolution less than 50uA. I don't have it.

qingz2004, 2018-06-17:
Same meter, tested with Sonoff S31 for 24 hours.

Tasmota with relay on: 0.0238kWh.
Stock  with relay on: 0.0010kWh.
Tasmota with relay on with 50 sleeptime: 0.0045kWh.
Tasmota with relay off: 24 hours, 0.0000kWh.

No matter what, the difference is obvious.

Frogmore42, 2018-06-17:
While I don't dispute the stock FW probably uses less energy than Tasmota, it also does less.  What I do dispute is the validity of your measurements.  What you are trying to measure is in the noise of the tool you are using to make the measurement.  At the lowest watt reading 0.1W, 24hr is 2.4WHr.  This is half of the value you got for Tasmota.  Now, one valid way to interpret this result is that Tasmota uses more than 4x the energy.  That is what your numbers say if you don't look at the details.  Another way to look at the results is that they are much closer to the same.  Here's how to interpret it that way.  Assume that stock is just under the value for 0.1W, so sometimes it gets read as 0.1 and other times it gets read as 0.  That will explain what you see for stock.  Now assume that Tasmota is right around the level for 0.2.  So most of the time it gets read as 0.2.  This explains the Tasmota results, but now the difference is about 2x (of a very small number).

I am not sure what your point is though.  Power efficiency is not generally a top priority for mains powered devices.  It is also very hard to do well.  If power efficiency is your top priority, you are free to use the stock FW.  If you have ideas on how to improve the power efficiency of Tasmota, please share them (and the code to implement them).  Generally, for an MCU design it would involve lots of little changes and a significant change in the structure of the software.  The basic thing is to make it narcoleptic, so the MCU and peripherals are in their lowest power states as much as possible.  Doing that correctly is not easy and I don't know how well the frameworks would support it

qingz2004, 2018-06-17:
Tasmota is great firmware with lots of great features. I flashed every single esp8266 device of mine with it. I'm not complaining it. Just try to ask for help to better understanding what is happening here. To see if there is a way to make it more energy efficiency.
I had a plan to be able to lock/unlock my car remotely, or be able to unlock it if my key is locked in the car. If it drains too much battery, I can't do it. Even with stock FW, it's not practical.

Frogmore42, 2018-06-17:
Sleep is the way to do that.  The esp8266 is not a good idea for battery operation.  It can be done, but it is not easy.

If you want low power suitable for use on battery, look here https://lowpowerlab.com

qingz2004, 2018-06-17:
Will do.
Thanks!

ascillato, 2018-06-21:
Hi,

If your issue is solved, please close it. Thanks! :+1: