arendst/Tasmota #2966: not a bug, but calibration on SI1145

gnkarn, 2018-06-10:
**Describe the bug**

UV index calculation is offset by  0.23 points

tested on node mcu 1.0 , 
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
using SY1145 , wait until  sun down , and the remaining uv indication is still non zero .

**Expected behavior**
indicate zero , on total darkness when sun is down .
**Screenshots**
[image]

**Additional context**
i will check the IR, and visible tomorrow .
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-06-10:
Hi,

Interesting. It should be zero? See this [link](https://www.quora.com/What-happens-to-uv-rays-at-night-when-there-is-no-sun)

ascillato, 2018-06-10:
Also see this https://www.allaboutcircuits.com/projects/how-to-perform-uv-monitoring-with-the-adafruit-si1145/

ascillato, 2018-06-12:
Hi, @gnkarn 

Any news about this?

gnkarn, 2018-06-12:
??  Don’t know what you mean. 
There is a problem with the values reported by the library , it is probably not a big deal as if it. Is only an offset it can be solved on the user side just by compensating with a substraction .

My intention was just a watch out .

I willl also report the visible and ir values, but it has been totally cloudy these days , so no new data so far .

ascillato, 2018-06-13:
Hi,

I was just asking if the value of 0.23 UV Index by night need to be zero or not. In the first link, they said that by night you will have few UV due to reflection from the atmosphere, so a little UV measurement could be ok ? I'm asking.
The second link shows a display and a UV Index sensor, and it is showing 0.02 Index inside a house.

So, that is my question. 0.23 UV Index by night is not Ok ? I don't know, that's why I'm asking you.

ascillato, 2018-06-21:
Hi. Any news on this? Or we just close this issue? Thanks.

gnkarn, 2018-06-21:
You can close it , I’m not able to compare the values against a “calibrated source”, as said in my opinion the values have an offset , that can be removed on the user side , probably is something that could be mentioned on the docs. Thanks

closed 2018-06-21 by gnkarn