arendst/Tasmota #2967: sonoff lights always on

arissaraiva, 2018-06-11:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used : 
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**
Hello everyone, I´m facing a problem that seems more like a hardaware than with tasmota. I was able to flash tasmota OTA to one basic sonoff, he´s working perfctly, I attached him to AC and he´s replying to my commands, but when I attached the bulb to test, he makes the sound of the toggled, he turn off the led green inside him, but the lights never fade, it keeps on alway, anyone could help me?

rgds

ascillato, 2018-06-11:
Hi,

Could you be so kind to complete the template to let us know a little more of your setup. Because, now I have to ask you which Hardware are you using (as states in the template) ? a Sonoff Basic?

If that is the case, may be a faulty relay? a wrong connection?

Please also provide some photos of your connection.

Thanks.

arissaraiva, 2018-06-11:
Tks for the quick reply, I'm using a sonoff basic, I attached to my socket just for test and put a gu10 led light on the output I dont have pictures right now, but a tried both ways input from AC power and also from the lamp and it gives me the same always, lights always on.

Frogmore42, 2018-06-11:
You probably have it wired incorrectly, since the relay is switching based on what you said.  Look at all the information available about how to wire it up, either here or at Itead, or just about anywhere.  The other option is the Sonoff is defective, but that is unlikely if you can here the relay switching.

arissaraiva, 2018-06-11:
[image]
Here´s a photo of the wiring, in the up part I attached the AC, hot on line and neutral on neutral, and the downside I attached the bulb. I was testing with a led bulb of 5w could be the lower voltage of the bulb? I´ll try today with a common e27.

Jason2866, 2018-06-11:
Please do not connect this wiring to AC again!! 
**It is DANGEROUS!!!!!!!!**

arissaraiva, 2018-06-11:
I know eletricity it´s dangerous but I´m using duck tapes for isolating the wires and at the end of the wire I have a power plug this is just a footage of before everything, this is just a test I´m not putting the wires direct on the power. I apreaciate more help on this

Jason2866, 2018-06-11:
Dont use wires drilled by hand with open isolations. You are scaring me and you really risk your life
Please use a full isolated wire and place the 2 wires (AC) in the connector N/L Input.
Your Lamp is on N/L Output with 2 wires. Thats all.

arissaraiva, 2018-06-11:
that´s what I´m doing and using wiring nuts or isolation duck tape for the wires, but still not working the lights kept on all the time, the relay makes sound but are not switching off the lamps.

Jason2866, 2018-06-11:
Maybe you have a defect device. I see on the case that you have got a short cut.

arissaraiva, 2018-06-11:
I bought another one will try, anyone has experience using led lights, like gu10 with low watts?

Frogmore42, 2018-06-11:
The photo indicates you should not be attempting this type of project, at least not before you spend a lot of time learning about the basics of electrical wiring safety.  The length of exposed wire at each of the connections is way too long.  The char/black spots on the case indicates the two exposed wires on both the input and the output touched at some point.  Given the behavior you are seeing, it is possible that the relay contacts are now fused.  It should have blown a fuse first.  As the others have said, your wiring is very unsafe and you are lucky to still be alive and that your house is still okay.

LED bulbs are no different then regular ones.  Your Sonoff is now defective and most likely due to the multiple shorts that likely occurred due to the wiring errors.

arissaraiva, 2018-06-11:
ok tks very much

closed 2018-06-11 by ascillato2