arendst/Tasmota #2969: BME 680 (IAQ) level

moneybag, 2018-06-11:
Thanks for providing the bme680 in generic.
Is it possible to show the IAQ level, not just the level in kOhms?
Does tasmota uses the newest firmware?
https://www.bosch-sensortec.com/bst/products/all_products/bsec

Chaot1, 2018-06-17:
I think that the complete integration is not working correct.
My Sensor readings are always changing. Not as described in datasheet with heating period and reading every x seconds.
Additionally i can watch a constantly rising resistance when i keep the website open.

moneybag, 2018-06-18:
thats my opinion, too. Lets wait and see, if someone corrects the problems. The price of the bme 680 is quite cheap.

arendst, 2018-07-04:
Rewrote the BME680 driver dropping Adafruit and now only using the latest Bosch driver v3.5.9. Saving 2k of code space.

With regards to IAQ BSEC library:

> Due to the current size of the BSEC library, upon compilation, you will receive an error: section ‘.text' will not fit in region ‘iram1_0_seg'. In order to solve this, you will need to modify the linker script and speciﬁcally deﬁne where the librarys hould be placed in memory. 

So I guess that's no option as Tasmota is already running low on ram.

Chaot1, 2018-07-04:
I think that the BME680 is a cheap sensor, but without propper integration it would be useless.
@arendst Did you ever hear from the MCU-811 or CJMCU-811?
It seems an eaqual Sensor.