arendst/Tasmota #2973: POW R2 on tasmota 5.14 reboot cycle

Smirlis, 2018-06-11:
I have Sonoff POW R2. Flashed with Tasmota 5.12, but on POW configuration doesn't show voltage/amperage/etc measurements.
Flashed with 5.14, but aftre switching to POW R2 module configuration it boot cycles. Terminal shows:

```
wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

00:00:00 APP: Set Baudrate to 4800
```

Managed to get some output on console:

```
00:00:00 APP: Set Baudrate to 4800
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_B1D551, GroupTopic sonoffs) Version 5.14.0-2_4_1
00:00:00 WIF: Connecting to AP1 GIOT in mode 11N as sonoff-5457...
20:26:16 WIF: Connected
20:26:16 DNS: Initialized
20:26:16 HTP: Web server active on sonoff-5457.local with IP address 192.168.1.162
20:26:17 MQT: Attempting connection...
20:26:17 MQT: Connected
20:26:17 MQT: tele/sonoff/LWT = Online (retained)
20:26:17 MQT: cmnd/sonoff/POWER = 
20:26:17 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Pow R2","Version":"5.14.0","FallbackTopic":"DVES_B1D551","GroupTopic":"sonoffs"}
20:26:17 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5457","IPAddress":"192.168.1.162"}
20:26:17 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software Watchdog"}
20:26:18 MQT: homeassistant/light/sonoff_1/config =  (retained)
20:26:18 MQT: homeassistant/switch/sonoff_1/config = {"name":"Sonoff","command_topic":"cmnd/sonoff/POWER","state_topic":"stat/sonoff/RESULT","value_template":"{{value_json.POWER}}","payload_off":"OFF","payload_on":"ON","availability_topic":"tele/sonoff/LWT","payload_available":"Online","payload_not_available":"Offline"} (retained)
20:26:18 MQT: homeassistant/light/sonoff_2/config =  (retained)
20:26:18 MQT: homeassistant/switch/sonoff_2/config =  (retained)
20:26:18 MQT: homeassistant/light/sonoff_3/config =  (retained)
20:26:18 MQT: homeassistant/switch/sonoff_3/config =  (retained)
20:26:18 MQT: homeassistant/light/sonoff_4/config =  (retained)
20:26:18 MQT: homeassistant/switch/sonoff_4/config =  (retained)
20:26:18 MQT: homeassistant/light/sonoff_5/config =  (retained)
20:26:18 MQT: homeassistant/switch/sonoff_5/config =  (retained)
20:26:18 MQT: homeassistant/light/sonoff_6/config =  (retained)
20:26:20 MQT: homeassistant/switch/sonoff_6/config =  (retained)
20:26:20 MQT: homeassistant/light/sonoff_7/config =  (retained)
20:26:20 MQT: homeassistant/switch/sonoff_7/config =  (retained)
20:26:20 MQT: homeassistant/light/sonoff_8/config =  (retained)
20:26:20 MQT: homeassistant/switch/sonoff_8/config =  (retained)
20:26:22 MQT: stat/sonoff/STATUS = {"Status":{"Module":43,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:26:22 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://intranet.lan/tasmota/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:00:12","StartupUTC":"2018-06-11T19:26:10","Sleep":0,"BootCount":66,"SaveCount":70,"SaveAddress":"F6000"}}
20:26:22 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-06-11T22:05:49","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
20:26:22 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"syslog.lan","LogPort":514,"SSId":["GIOT","none"],"TelePeriod":300,"SetOption":["000A8009","55818000"]}}
20:26:22 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":513,"Free":488,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
20:26:22 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5457","IPAddress":"192.168.1.162","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:F3:EB:B1:D5:51","Webserver":2,"WifiConfig":1}}
20:26:26 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B1D551","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:26:26 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Jun 11 19:26:26 2018","Local":"Mon Jun 11 20:26:26 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:46","Sunset":"20:53"}}
20:26:26 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:26:27 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-06-11T20:26:26","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
20:26:27 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-06-11T20:26:27","Uptime":"0T00:00:17","Vcc":3.636,"POWER":"OFF","Wifi":{"AP":1,"SSId":"GIOT","RSSI":88,"APMac":"00:16:01:27:63:55"}}}
20:26:29 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
20:26:29 MQT: stat/sonoff/POWER = OFF
20:26:33 MQT: stat/sonoff/STATUS = {"Status":{"Module":43,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:26:33 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://intranet.lan/tasmota/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:00:23","StartupUTC":"2018-06-11T19:26:10","Sleep":0,"BootCount":66,"SaveCount":70,"SaveAddress":"F6000"}}
20:26:33 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-06-11T22:05:49","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
20:26:33 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"syslog.lan","LogPort":514,"SSId":["GIOT","none"],"TelePeriod":300,"SetOption":["000A8009","55818000"]}}
20:26:33 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":513,"Free":488,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
20:26:36 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5457","IPAddress":"192.168.1.162","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:F3:EB:B1:D5:51","Webserver":2,"WifiConfig":1}}
20:26:36 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B1D551","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:26:36 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Jun 11 19:26:36 2018","Local":"Mon Jun 11 20:26:36 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:46","Sunset":"20:53"}}
20:26:36 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:26:36 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-06-11T20:26:36","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
20:26:36 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-06-11T20:26:36","Uptime":"0T00:00:26","Vcc":3.634,"POWER":"OFF","Wifi":{"AP":1,"SSId":"GIOT","RSSI":88,"APMac":"00:16:01:27:63:55"}}}
```

What should I check?

Thanks.

Frogmore42, 2018-06-12:
Looks like you probably have a lot of stuff on your network and a lot of retained messages.  The heap size looks quite low, which means it might be running out of memory.  Is this your only device?

The POW doesn't measure unless the relay is on.  The power is off in the log, so having 0 values is expected.

Using the POW module type when you actually have a POW R2 will not work, but make sure you look at the wiki for how to tell the difference between the two versions.

You are using Core 2.4.1, did you compile the code yourself or load a pre-built bin?

Smirlis, 2018-06-12:
I have 5 sonoffs and couple more apps which interact via MQTT.

I don't know where these coming from:
```
20:26:20 MQT: homeassistant/switch/sonoff_7/config =  (retained)
20:26:20 MQT: homeassistant/light/sonoff_8/config =  (retained)
```
Could it be from homeassistant discovery functionality?

Sonoff seems stable while in Basic module configuration, so I guess it is not MQTT related, but rather POW related.

Where do you see the heap size in the log?

I compiled it with Arduino 1.8.5.
I'll try pre-built version later if fail/DIY_fun ratio increases significantly.

znanev, 2018-06-14:
@Smirlis Look at the output of the command `SetOption19`. If it says 'ON' - that is what publishes those messages.

To turn Home Assistant automatic discovery functionality off, execute the following in the Console menu:

`SetOption19 0`

alfista2600, 2018-06-17:
I have a similar problem... compiled with 1.8.5 on a Mac.

Neither of my first two SONOFF POW's are showing data. Here are the versions:

Self-compiled:

> Program Version | 5.14.0
> -- | --
> 2018-06-16T10:44:00
> 2_4_1/2.2.1(cfd48f3)
> 

OTA download over self-compiled:

> Program Version | 5.14.0
> -- | --
> 2018-05-15T15:29:54
> 2_3_0/1.5.3(aec24ac9)
> 
>

Frogmore42, 2018-06-17:
There are to types of POW devices.  You have to select the correct one in the module UI to match the type you have.

alfista2600, 2018-06-18:
Thanks - the board said Rev2 so I assumed it was, but changing it to the first POW option fixed the issue.

Smirlis, 2018-06-19:
For me the fix was compiling without NTP configuration. Setting up NTP at later stages work without problems. Still not sure about it.

closed 2018-06-19 by Smirlis