arendst/Tasmota #2974: MQTT intermittently goes offline and doesn't reconnect.

nosaj66au, 2018-06-12:
Firmware 6.0.0a (note the Wemos are the only unit currently at 6.0a the others are at 5.12.0i
Configuration is 4 switches and 4 relays
All set to switchmode[x] = 2 and setoption13 = "on"

The network has approximate 14 devices 8 * H801 , 1 * Sonoff TH16 and 5 * Wemos D1 R2, the automation software is Node Red on a PI with a Mosquito broker. After a period of intermittent  time but greater than 4 hours the MQTT connection goes offline as reported by the LWT on the broker. This only occurs to 1 wemos D1 R2 device at a time and has happens 4 times over the passed 3 days.  To date only 2 of the 5 Wemos devices has had the issues. While occuring the device is still on the network and the network diag. does not indicate a Wifi outage with a signal strength is 56%.  The device doesn’t appear to attempting to re connect to the broker from the Console logs. However connects fine after a reboot from the web page.    

STATUS 0 OUTPUT HERE
08:23:57 MQT: stat/Switch_103/STATUS = {"Status":{"Module":18,"FriendlyName":["Switch_103","Sonoff2","Sonoff3","Sonoff4"],"Topic":"Switch_103","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

**To Reproduce**
Extended period of time. I have not been able to tie it to and event like internal or external action 
No rules or Timers are being used. 
 
**Expected behavior**
MQTT not to disconnect or to reconnect if it does.

Jason

Frogmore42, 2018-06-12:
You didn't include complete status 0

There are many reasons why something can fail, and your failure rate is actually fairly low, which means it will take a fair amount of effort to figure it out.  You will need to capture a console log (probably with a higher level of logging) when the problem happens to have any hope of figuring out the cause.

I have one NodeMCU device that regularly restarts.  I haven't looked at it closely enough to know exactly why, but I suspect the power regulator on it is bad and it sometimes drops the voltage enough that the esp8266 resets.

Since you are using different hardware, running different firmware, with likely different versions of the core libraries, you are going to have to make a detailed list of all of that and keep track of which ones are failing to see if there is any pattern.

nosaj66au, 2018-06-12:
I have been using "http://192.168.6.103/cm?cmnd=restart 1" to restart it manually from NR Dashboard when it happens, but I  can set Node Red to do this automatically. 
Is the a command to just reset the MQTT connection which is less brutal a full reboot ? . Is there anything wrong with just auto rebooting?

Frogmore42, 2018-06-12:
It is not normal for Tasmota to lose connection and not retry.  But, if you are happy with the workaround of "reboot it regularly", go for it.  There is always more than one way to "solve" a problem.

If you actually want help in figuring out what the real problem is, you will need to figure out how to get the web/serial log and especially the complete status 0
This will show exactly what version of the software and core components you are using.

To figure out what is really happening will require some effort on your part, since it is inconsistent behavior.

nosaj66au, 2018-06-12:
I understand and will get the data however heading away for a few day and need to keep the wife happy while I'm away. Auto reset is only a short trem sollution.
Thanks 
Jason

ascillato, 2018-06-12:
Hi Jason

If your issue is solved, please close it. Thanks :+1:

nosaj66au, 2018-06-13:
This is odd I have changed from switchmode 2 to switchmode 3 for all 4 switches to lower MQTT outputs numbers. Now the issues has stopped happening. ????

closed 2018-06-13 by nosaj66au