arendst/Tasmota #2975: Corrupted picture in wiki

jnt2007, 2018-06-12:
**There is corrupted picture on Sonoff Basic wiki page**

**To Reproduce**
_Just open you wiki page https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic_

**Expected behavior**
_All pictures looks good_

**Screenshots**
[image]

**Just repalce corrupted picture:)**

ascillato2, 2018-06-12:
It is working now. Thanks.

closed 2018-06-12 by ascillato2

jnt2007, 2018-06-12:
@ascillato2 Excuse me, but no, it's not working now. I tried in Chome browser on Ubuntu and on my iPhone. Picture still corrupted...

ascillato, 2018-06-12:
[image]

jnt2007, 2018-06-12:
So strange, because I can't see picture complete in a different browsers and devices (deleting cookies and cache does not effect). Maybe it's a GitHub issue with UPLINK to my location.

Can you reupload picture? It mitght be a key to resolve problem for me...

ascillato, 2018-07-23:
Wiki Edited.

Please, check again. Thanks

jnt2007, 2018-07-23:
No changes...(