arendst/Tasmota #2976: [Question] Solder an extra LED on Sonoff SV

BrutalBirdie, 2018-06-12:
Hello guys, I wanna solder an extra LED to the GPIO4, GPIO5 or GPIO14, but so far I had no success.
I want the LED to turn on the same way as the on-board LED.
"Why tho?" you might ask, well the SMD LED is broken and I don't have an a spare SMD LED.
Is there a way to solder and extra LED to the GPIOs?

Thanks for reading and looking forward to a solution :)

znanev, 2018-06-12:
Hi,

I see no problem soldering an additional LED to some of the free GPIO pins.

Solder LED's anode to +3.3V, solder one end of a 1K resistor to LED's cathode, and the other end of the resistor to your chosen GPIO pin.

Then you'll probably have to configure your module as Generic and set the appropriate GPIO pins functions.

BrutalBirdie, 2018-06-13:
`configure your module as Generic and set the appropriate GPIO pins functions`
Uhmmm... can you maybe help me with that part ? :rofl:

BrutalBirdie, 2018-06-13:
Okay figured it out myself, with a bit of reading and studying the schematics.
[image]

closed 2018-06-13 by BrutalBirdie

znanev, 2018-06-13:
@BrutalBirdie Good job! I was about to post you a similar screenshot, but I'm glad that you figured it out yourself :)